tonyp7/esp32-wifi-manager #2: Enhancement ideas: Select AP-name, AP-channel, STA-only, WiFi-Power save & Websockets

FredrikFornstad, 2017-12-27:
Thanks for a great application!
I have a few more needs myself and I am trying to find my way through your code but have not yet managed to do what I want:
1. From the main web page bring up a new page "Settings" (for instance by clicking the "WiFi header")
2. In the Settings page be able to:
  a. Set (and store) a new name for the AP
  b. Set (and store) a new WiFi channel for the AP
  c. Select (and store) if the AP should be turned off whenever esp32 is successfully connected as a WiFi client (i.e. switch to "STA" mode instead of "APSTA" whenever connected to a WiFi router)
  d. Select (and store) if WiFi Power save mode should be enabled when in client mode

3. In my own application, I think I will need to use websockets instead of regular Json pulls due to latency. If WiFi-manager could be changed to use websockets instead of regular Json pulls, then I guess it would make WiFi-manager:
  a. acting more in "real time" when scanning for WiFi AP's or when signal strengts varies etc
  b. easier to integrate with applications that need to read/control I/O pins in "real time" (like my app)

Any chance that you are already working on similar additions to WiFi-manager (to save me hours of work :-) )?

tonyp7, 2017-12-28:
Hi Fredrik,

Well thanks for the interest you're showing in the app! So getting to your points:

1 & 2:
_Set (and store) a new name for the AP_ 
Definetely something possible.

_Set (and store) a new WiFi channel for the AP_
This one is tricky. Because there's only one radio chip on the esp32, your AP must be on the same channel as your wifi it is connected to. So even if say, you save 1 and your home wifi is on 6, the number "1" becomes sort of meaningless as soon as your esp32 connects to your home wifi.
An elegant solution would be to automatically update and start the softAP channel number using the last wifi connected to. Would that make sense?

_Select (and store) if the AP should be turned off whenever esp32 is successfully connected as a WiFi client (i.e. switch to "STA" mode instead of "APSTA" whenever connected to a WiFi router)_
I thought about this one and it's something I would like to add yes. Makes the whole esp32 completely disappear once the pairing is done!

_Select (and store) if WiFi Power save mode should be enabled when in client mode_
What I'm working on is AC powered so I will admit I have not given any thought at all about power saving. I need to look at the doc first see what it does!

3:
I created this app because I am developing an embedded system that needs the internet. I do not want to have to reflash it or anything any time my internet breaks, or I change my wifi router's password or anything else. But it was never intended to be able to interact with real time embedded systems. It connects to your wifi and that's all there is to it. 
I would suggest you completely decorrelate your app from the wifi manager by using your own web server that can run on sockets on the port 8080 for instance (or move esp-wifi-manager to 8080) because that is a change that is very unlikely at this point!

One thing that is missing today and I am working on is an auto-reconnect feature. Today if your router dies or anything else of this nature the esp32 is kind of lost. So most likely in the near future:
1. Auto-reconnect
2. Setting pages
3. Select AP name
4. Choose to disable softAP
5. _Maybe_ the softAP channel?

FredrikFornstad, 2017-12-29:
Great!
The idea with channel number is in cases where the esp(s) ends up in an environment  with no router and the user would like have them on separate channels as softAPs to keep interference down. As you say, whenever esp connect to a router it will switch to that frequency which is fine with me. So no need to remember last station channel...

I agree that the autoconnect feature is much needed. (I actually thought it was there already...). In case the softAP has been turned off and the router is lost, it is ok with me to wait until next reboot before turning on the softAP again (while still trying to connect to the router without user intervention)

zafrirron, 2018-01-14:
Thanks for a great application! Also from myself,

Looking into your great program and while asking my self the same questions on this thread, I did few modifications you may want to consider.

1. Yes -  changed to STA only once connected.
2.  Remove - wifi manager's web server task once connected as STA and add option to disconnect by external signal (personal web server/AWS thing/Blynk app), or unable to connect to the saved STA, reason is that once I deploy this on a wifi location I will rarely change the wifi settings. in my case I've set menuconfig options to select wifi manager configuration.
3. I would recommend using ESP_LOG(I/W/E) instead of printf for debug.
4. Since most application need to know that wifi connected I suggest to move wifi_manager_event_group to header file and and give other applications/tasks to wait on signal for connection (AWS IOT / Blynk in my case).

thanks again for a great work! using your work got me closer to my project goal, to get a generic lightweight ESP32 IOT framework that allow easy deployment and cloud connectivity for any ESP application. were most configuration done on kconfig.

SERIDJ, 2020-08-05:
hi, please i want management of my parameter application (wifi/ethernet and mqtt param) through a web portal how i can do that now iam using sdkconfig file 

[image]