tonyp7/esp32-wifi-manager #6: MQTT

reshuler, 2018-03-15:
Hi,

Thanks again for a nice program. 

 I've run into an issue combining the wifi-manager with MQTT. If I use a the MQTT app (task) separately (stand alone) it is able to publish messages okay to a remote host.  However, if I use MQTT app (task) with wifi-manager, the MQTT app is not able to connect to the remote host to publish messages.  I've gone through the wifi-manager code, and tested different options.  So far I've not found a solution to why the MQTT app will not work along aside the wifi-manager.  

I have been able to use the wifi-manager to select an access point and store the access point credentials to NVM.  Then on a reboot, I check to see if there are access points credential in the NVM, If there are, I used those credentials to attached to the access point and start the MQTT app.  I don't start the wifi-manager up in this case.  This configuration works okay.

Have you run see this issue?  Do you have any recommendations on combining wifi-manager with MQTT?

Thanks,
Robert

beaubrewer, 2018-03-15:
Robert,

I suggest you try something that needs to resolve a hostname without using
MQTT and see if that works. I have issues resolving DNS when the ESP32 is
in AP or AP/ST mode.

This may not have anything to do with the wifi manager code and more to do
with DNS resolution on the ESP32.

Just a thought.

reshuler, 2018-03-15:
Got it. Good suggestion.  I’ll let you know what I’m able to do.

I seen issues around DNS before with the ESP32.

Thanks,
Robert

tonyp7, 2018-03-18:
Hi Robert, 

Could you care to share your code? I'm not sure I could see where the wifi manager could interact with MQTT!

Thanks

canhchimmcodon, 2019-03-06:
Any feedback about this problem ? I have a project and I got an error when implemting MQTT with wifi manager. It is E (21301) MQTT_CLIENT: Error transport connect

anupam19, 2019-03-26:
Hi @reshuler were able to resolve it ? Please let me know. I am also having same issue.

ojousima, 2019-03-27:
I modified this project and ble2mqtt to be ESP-components and use them together to relay data to MQTT broker. You can check the code and submodules out [here] (https://github.com/ojousima/ojousima.esp32_blegw.c) if you're interested, I've not had problems with MQTT.

anupam19, 2019-03-30:
Hi @ojousima  I tried with your code but still mqtt fails with same error. I tried with different networks and also local mqtt broker I had tried but no luck

closed 2020-07-03 by tonyp7