tonyp7/esp32-wifi-manager #11: Anyway to use a hostname instead of an IP?

hchaudhary1, 2018-05-10:
Instead of going to http://192.168.1.1, is it possible instead to go to "http://myesp.config". Is that possible?

tonyp7, 2018-05-11:
Hi hchaudhary1, that would not be something you can solve with the code. You would have to add an entry into your DNS server so that mysesp.config gets you back to 192.168.1.1.

You can also buy your own domain and have it send you back to 192.168.1.1. 

There is mDNS but it has its own set of limitations.

hchaudhary1, 2018-05-11:
well, most routers in hotels and coffee shops will automatically force my laptop to open a page with a non-public DNS name. I never need to know the IP of the router...

Here is a Wikipedia article I found. I will research this more. It should be possible.
https://en.wikipedia.org/wiki/Captive_portal

lukecyca, 2019-01-03:
I have forked this and added captive portal functionality, but it currently only works on macOS and iOS clients, due to a very hacky DNS implementation that can only parse the very specific DNS request that I sniffed on the network during testing.

There may now be a more generalized DNS server implementation that can parse any DNS query and resolve it to 192.168.1.1.

https://github.com/craftmetrics/esp32-wifi-manager
https://github.com/craftmetrics/esp32-dns-server

hchaudhary1, 2019-01-14:
ESP team may support the captive portal ref:
https://github.com/espressif/esp-idf/issues/2723

tonyp7, 2019-05-11:
The current version now automatically opens a page for you, similarly to a hotel wifi ("captive portal"). Try it out!

closed 2019-05-11 by tonyp7