tonyp7/esp32-wifi-manager #21: wifi manager NVS does not work in dual CPU mode on esp32

snahmad, 2018-09-05:
Hi,

Can you test your wifi manager start and stop in freeRTOS dual core mode.

Thanks,
Naeem

snahmad, 2018-09-07:
NVS is now working but wifi_manager_lock_json_buffer crash when running on dual core mode and running  
wifi manager in separate task schedule to run on cpu_1. Can you try yourself please.

snahmad, 2018-09-08:
You wifi manager did not deinit calls before destroy I put now it is working.

I will share code later.

snahmad, 2018-09-11:
ESP_ERROR_CHECK(esp_wifi_stop() );
    ESP_LOGI(TAG, "wifi_init_sta deinit sta");
    ESP_ERROR_CHECK(esp_wifi_deinit() );

I added these deinit before exit 

void wifi_manager( void * pvParameters )

tonyp7, 2018-09-11:
I don't think you can stop wifi abruptly without properly cleaning up objects and memory. 

I created wifi_manager_destroy to keep track of the allocated object and because it is good practice but realistically today there is no easy way to stop the manager "cleanly" this should be an added feature.

snahmad, 2018-09-11:
It is working for me before esp_restart. otherwise it get heap corruption

snahmad, 2018-09-18:
Hi tony,

I start with default hard code name for DEFAULT_AP_SSID and DEFAULT_AP_PASSWORD.

How I need to stop wifi and change SSID and password programmatically either via serial or over the air using Http post comands.

Is it possible. I guess I need to stop and restart wifi after setting new ssid and password.

snahmad, 2018-10-05:
> I don't think you can stop wifi abruptly without properly cleaning up objects and memory.
> 
> I created wifi_manager_destroy to keep track of the allocated object and because it is good practice but realistically today there is no easy way to stop the manager "cleanly" this should be an added feature.

It is working in latest verison = 3.2. I try with version =3.1. got crash.

closed 2019-05-11 by tonyp7