tonyp7/esp32-wifi-manager #31: select aws_iot_connection network

zafersn, 2019-02-14:
Hi. I'm trying to connect to aws_iot, but I get "aws_iot: failed! mbedtls_net_connect returned -0x52" error. This error may be a network connection problem. The point I'm curious about is how the network configuration of the aws configuration is trying to connect and how I can select it.
I'm using aws embedded c sdk.

and 

I call aws_set_event_start() in the following code block.

```
		uxBits = xEventGroupWaitBits(wifi_manager_event_group, WIFI_MANAGER_WIFI_CONNECTED_BIT | WIFI_MANAGER_STA_DISCONNECT_BIT, pdFALSE, pdFALSE, portMAX_DELAY );

			if(uxBits & (WIFI_MANAGER_WIFI_CONNECTED_BIT | WIFI_MANAGER_STA_DISCONNECT_BIT)){

				/* Update the json regardless of connection status.
				 * If connection was succesful an IP will get assigned.
				 * If the connection attempt is failed we mark it as a failed connection attempt
				 * as it is important for the front end app to distinguish failed attempt to
				 * regular disconnects
				 */
				if(wifi_manager_lock_json_buffer( portMAX_DELAY )){

					/* only save the config if the connection was successful! */
					if(uxBits & WIFI_MANAGER_WIFI_CONNECTED_BIT){

						/* generate the connection info with success */
						wifi_manager_generate_ip_info_json( UPDATE_CONNECTION_OK );

						/* save wifi config in NVS */
						wifi_manager_save_sta_config();
						#if WIFI_MANAGER_DEBUG
									printf("aws_start_Bit\n");
						#endif
						aws_set_event_start();
					}
					else{

						/* failed attempt to connect regardles of the reason */
						wifi_manager_generate_ip_info_json( UPDATE_FAILED_ATTEMPT );

						/* otherwise: reset the config */
						memset(wifi_manager_config_sta, 0x00, sizeof(wifi_config_t));
					}
					wifi_manager_unlock_json_buffer();
				}
				else{
					/* Even if someone were to furiously refresh a web resource that needs the json mutex,
					 * it seems impossible that this thread cannot obtain the mutex. Abort here is reasonnable.
					 */
					abort();
				}
			}
```

**subscribe_publish_sample.c**

```

/*
 * Copyright 2010-2015 Amazon.com, Inc. or its affiliates. All Rights Reserved.
 * Additions Copyright 2016 Espressif Systems (Shanghai) PTE LTD
 *
 * Licensed under the Apache License, Version 2.0 (the "License").
 * You may not use this file except in compliance with the License.
 * A copy of the License is located at
 *
 *  http://aws.amazon.com/apache2.0
 *
 * or in the "license" file accompanying this file. This file is distributed
 * on an "AS IS" BASIS, WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either
 * express or implied. See the License for the specific language governing
 * permissions and limitations under the License.
[... 448 lines cut ...]
//        rc = aws_iot_mqtt_publish(&client, TOPIC, TOPIC_LEN, &paramsQOS1);
//        if (rc == MQTT_REQUEST_TIMEOUT_ERROR) {
//            ESP_LOGW(TAG, "QOS1 publish ack not received.");
//            rc = SUCCESS;
//        }
//    }
//
//    ESP_LOGE(TAG, "An error occurred in the main loop.");
//    abort();
//}

void aws_iot_abort(void)
{
#if WIFI_MANAGER_DEBUG
	 //ESP_LOGE(TAG, "An error occurred in the main loop.");
	 printf("An error occurred in the main loop.");
#endif
	 abort();
}

/*****************************************************************************/
/* END OF FILE */
/*****************************************************************************/

```

tonyp7, 2019-03-11:
This does not seem to be a valid issue for the wifi manager

closed 2019-03-11 by tonyp7

zafersn, 2019-03-11:
I think it is about wifi manager. Because.

 I've read and applied a solution for the "mbedtls_net_connect returned -0x52" error.
https://stackoverflow.com/a/46749602

It is said that the problem may be the static IP configuration and the missing or incorrect setting of the DNS settings. When I examined the wifimanager configuration, I observed that there was a static IP configuration available.

I have no knowledge about network configuration, so I tried to switch off and try static ip configurations in wifi configuration settings in wifimanager.c and the result was successful. I am no longer receiving the error "aws_iot: failed! Mbedtls_net_connect returned -0x52" . But now after receiving the message "autoreconnect ... Subscribing ... Subscribing Success", I get an error like "MQTT DisconnectAuto Reconnect is Enabled," when I wait before publishing a message. If I want to post a message, I get an error such as "E (58050) aws_iot: failed! Mbedtls_ssl_write returned -0x50".

How do you think this static ip and dns configuration should be?
static ip and dns configuration (delete for this project) how it solved this error: "aws_iot: failed! Mbedtls_net_connect returned -0x52" .