tonyp7/esp32-wifi-manager #36: Forever 4

app-js, 2019-03-20:
http_server.c line 169

`if ((sizeof(host) > 0) && !strstr(host, DEFAULT_AP_IP)) {`

should be

`if (host && !strstr(host, DEFAULT_AP_IP)) {`
or
`if (lenH > 0 && !strstr(host, DEFAULT_AP_IP)) {`

And if Wifi is running in STA mode, then it can't be accessed via the network.

tonyp7, 2019-03-20:
Looks about right, I mark it as bug
Thanks for the report

tonyp7, 2019-05-11:
This is fixed in the latest version.

Now you can access the http server in STA mode. This wasn't an easy fix by any mean, this is what the code looks like now:

`/* captive portal functionality: redirect to access point IP for HOST that are not the access point IP OR the STA IP */
			int lenH = 0;
			char *host = http_server_get_header(save_ptr, "Host: ", &lenH);
			/* determine if Host is from the STA IP address */
			wifi_manager_lock_sta_ip_string(portMAX_DELAY);
			bool access_from_sta_ip = lenH > 0?strstr(host, wifi_manager_get_sta_ip_string()):false;
			wifi_manager_unlock_sta_ip_string();

			if (lenH > 0 && !strstr(host, DEFAULT_AP_IP) && !access_from_sta_ip) {`

closed 2019-05-11 by tonyp7