tonyp7/esp32-wifi-manager #37: Fast scan and its implication

snahmad, 2019-03-22:
// configure the softAP and start it */
	wifi_config_t ap_config;
	ap_config.ap.ssid_len = 0;
	ap_config.ap.channel = __wifi_settings.ap_channel;
	ap_config.ap.authmode = (wifi_auth_mode_t)__wifi_settings.ap_authmode;
	ap_config.ap.ssid_hidden = __wifi_settings.ap_ssid_hidden;
	ap_config.ap.max_connection = AP_MAX_CONNECTIONS;
	ap_config.ap.beacon_interval = AP_BEACON_INTERVAL;
	memcpy(ap_config.ap.ssid, __wifi_settings.ap_ssid, sizeof(__wifi_settings.ap_ssid));
	memcpy(ap_config.ap.password, __wifi_settings.ap_pwd, sizeof(__wifi_settings.ap_pwd));
	ap_config.sta.scan_method = WIFI_FAST_SCAN;

	ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &ap_config));
	ESP_ERROR_CHECK(esp_wifi_start());

tonyp7, 2019-03-25:
I don't understand what you're trying to report?

snahmad, 2019-03-25:
What is purpose of WIFI_FAST_SCAN as compare all scan.

tonyp7, 2019-03-26:
This has nothing to do with the manager, you can read the doc here:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html

closed 2019-03-26 by tonyp7