tonyp7/esp32-wifi-manager #39: Errors on ESP-IDF 4.0

rokrodic, 2019-04-16:
Hi. I have downloaded your code and try to to run it. This is what I get:

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:8580
load:0x40078000,len:12064
load:0x40080400,len:7116
entry 0x400807a0
D (64) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (35) boot: ESP-IDF v4.0-dev-278-gdaee3dfa5 2nd stage bootloader
I (36) boot: compile time 23:04:18
D (36) boot: Enabling RTCWDT(9000 ms)
I (49) boot: Enabling RNG early entropy source...
D (49) boot: magic e9
D (49) boot: segments 04
D (51) boot: spi_mode 02
D (53) boot: spi_speed 0f
D (56) boot: spi_size 02
I (58) boot: SPI Speed      : 80MHz
I (63) boot: SPI Mode       : DIO
I (67) boot: SPI Flash Size : 4MB
D (71) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (80) boot: mapped partition table 0x8000 at 0x3f408000
D (85) flash_parts: partition table verified, 9 entries
I (90) boot: Partition Table:
I (94) boot: ## Label            Usage          Type ST Offset   Length
D (101) boot: load partition table entry 0x3f408000
D (106) boot: type=1 subtype=2
I (109) boot:  0 nvs              WiFi data        01 02 00009000 00005000
D (117) boot: load partition table entry 0x3f408020
D (122) boot: type=1 subtype=0
I (125) boot:  1 otadata          OTA data         01 00 0000e000 00002000
D (132) boot: load partition table entry 0x3f408040
D (137) boot: type=0 subtype=0
I (140) boot:  2 factory          factory app      00 00 00010000 00120000
D (148) boot: load partition table entry 0x3f408060
D (153) boot: type=0 subtype=10
I (156) boot:  3 app0             OTA app          00 10 00130000 00100000
D (164) boot: load partition table entry 0x3f408080
D (169) boot: type=0 subtype=11
I (172) boot:  4 app1             OTA app          00 11 00230000 00100000
D (179) boot: load partition table entry 0x3f4080a0
D (184) boot: type=1 subtype=99
I (188) boot:  5 eeprom           Unknown data     01 99 00330000 00001000
D (195) boot: load partition table entry 0x3f4080c0
D (200) boot: type=1 subtype=82
I (203) boot:  6 storage          Unknown data     01 82 00331000 00060000
D (211) boot: load partition table entry 0x3f4080e0
D (216) boot: type=1 subtype=4
I (219) boot:  7 nvs_key          NVS keys         01 04 00391000 00001000
I (227) boot: End of partition table
D (231) boot: OTA data offset 0xe000
D (234) bootloader_flash: mmu set paddr=00000000 count=1 size=2000 src_addr=e000 src_addr_aligned=0
D (244) boot: otadata[0]: sequence values 0xffffffff
D (249) boot: otadata[1]: sequence values 0xffffffff
D (254) boot: OTA sequence numbers both empty (all-0xFF) or partition table does not have bootable ota_apps (app_count=2)
I (265) boot: Defaulting to factory image
D (269) boot: Trying partition index -1 offs 0x10000 size 0x120000
D (276) esp_image: reading image header @ 0x10000
D (280) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (287) esp_image: image header: 0xe9 0x06 0x02 0x02 400810c4
I (293) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2687c (157820) map
D (302) bootloader_flash: mmu set paddr=00010000 count=3 size=2687c src_addr=10020 src_addr_aligned=10000
D (358) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)
[... 180 lines cut ...]
D (418) nvs: nvs_get ap.phybw 1
D (428) nvs: nvs_get ap.sndchan 1
D (428) nvs: nvs_get lorate 1
D (428) nvs: nvs_set_blob sta.mac 6
D (448) nvs: nvs_set_blob ap.mac 6
ESP_ERROR_CHECK failed: esp_err_t 0xb (ERROR) at 0x40088008
0x40088008: _esp_error_check_failed at D:/msys32/home/Akoya/esp/esp-idf/components/esp32/panic.c:721

file: "D:/msys32/home/AKOYA/esp/wifi_manager/main/wifi_manager.c" line 580
func: wifi_manager
expression: esp_wifi_set_config(WIFI_IF_AP, &ap_config)

ELF file SHA256: 3deb2d4cdfbba3d6f9dd637f484a0e62cb697f04a7c2f9b1445014692005d6af

Backtrace: 0x40087b9c:0x3ffbef60 0x4008800b:0x3ffbef80 0x400d3c76:0x3ffbefa0 0x4008d161:0x3ffbf110
0x40087b9c: invoke_abort at D:/msys32/home/Akoya/esp/esp-idf/components/esp32/panic.c:716

0x4008800b: _esp_error_check_failed at D:/msys32/home/Akoya/esp/esp-idf/components/esp32/panic.c:722

0x400d3c76: wifi_manager at D:/msys32/home/AKOYA/esp/wifi_manager/main/wifi_manager.c:580 (discriminator 1)

0x4008d161: vPortTaskWrapper at D:/msys32/home/Akoya/esp/esp-idf/components/freertos/port.c:403
`

Any idea why it reboots at /wifi_manager.c" line 580?

tonyp7, 2019-04-17:
esp-idf v4.0? Do you mean 3.2?

rokrodic, 2019-04-17:
I believe it's 4.0. From the output above:
I (35) boot: ESP-IDF v4.0-dev-278-gdaee3dfa5 2nd stage bootloader

rokrodic, 2019-04-17:
Hi. I have retried many times and finally got it working. The problem were causing both bootsector and ESP-IDF debug levels, which were turned to the full. Now I have them set to "warnings" and the code works...
Closing Issue.

closed 2019-04-17 by rokrodic