tonyp7/esp32-wifi-manager #40: Time to execute esp_wifi_scan_start

alexanderturner, 2019-05-05:
Hey there,
I'm writing an application on top of wifi-manager that extends the web server such to support my own API. In development I noticed that the esp_wifi_scan_start appears to be stealing 2.5s of CPU time every time it runs (which if often). My next course of action is to disable scanning once the unit enters STA however I was wondering if it was something you've come across and if there are more event driven alternatives?

Alex

alexanderturner, 2019-05-05:
Digging a bit deeper into this to force the IDF functions event-_drivenness_ with a new arg in ```wifi_manager_event_handler``` and passing false to the blocking attribute. It appears to be an issue with esp-idf blocking the thread when running the scan. Perhaps there should be a timer arg for the wifi scan instead of it being called every time jquery call it.

tonyp7, 2019-05-09:
Helo Alex,

You can indeed set it to false but this brings a whole lot of new issues with thread synchronization. However, it is true blocking the thread is not ideal.

tonyp7, 2019-05-10:
This is fixed in branch 2.0

tonyp7, 2019-05-11:
Fixed in latest master trunk -- scans are now non blocking

closed 2019-05-11 by tonyp7

alexanderturner, 2019-05-11:
Thanks for this Tony! I should have jumped in with a PR sooner but awesome to see this being actively maintained. I had set the blocking field to false and actioned the update task on in the event handler but whilst it's now 'non blocking' it's still blocking the thread for similar time when it's executing the scan (which is pretty frequently when the jquery script is running).

I'm working on project which could benefit from RTOS queues - are you keeping that effort in a working branch?

tonyp7, 2019-05-11:
I merged my dev branch into the master so you can simply get the master :)

Queue process starts here:
https://github.com/tonyp7/esp32-wifi-manager/blob/1a710dabcb9e7f7e0f3a8bb68fd0048ff4681908/wifi_manager/main/wifi_manager.c#L730

alexanderturner, 2019-05-11:
Ah awesome :) Thanks. Would be awesome to have a build status for master :)