tonyp7/esp32-wifi-manager #43: Suggestions

JoaoLopesF, 2019-05-19:
First of all, congrats to excellent coding on ESP-IDF, this is awesome ;-).

I have a few suggestions:

- Made this as component for easy deployment
- The password with confirmation (2nd entry for it)
- When finished, put one button to close the captive portal

Best regards

ojousima, 2019-05-20:
👍 for making the project as a component. 
I structured this project into a component a while ago, https://github.com/ojousima/esp32-wifi-manager, however I'll only update it when I happen to be working on my main project. I'd be happy to collaborate somehow in an effort of making maintained component of the project.

JoaoLopesF, 2019-05-20:
Hi @ojousima I will look your repo, congrats for you work
Regards

tonyp7, 2019-05-21:
Hello guys,
Can someone point me to the right direction as to what "making it into a component" entails? Is that similar to a DLL?

@JoaoLopesF on the remaining two comments:

- No WiFi portal out there asks you to reconfirm password. I feel like it would not be a very good UI move.
- I am planning to implement a ~30s timer once the wifi is connected that will automatically shutdown the AP. It was supposed to be part of release 2.0 but didn't make the cut.

ojousima, 2019-05-21:
> Can someone point me to the right direction as to what "making it into a component" entails? Is that similar to a DLL?

It's a bit different, while DLLs are precompiled the ESP components have the source and will be compiled  along with the rest of the project. [ESP build system documentation](https://docs.espressif.com/projects/esp-idf/en/stable/api-guides/build-system.html?highlight=component) has more details. 

I'm a beginner on ESP myself, but according to docs
> A component is any directory that contains a component.mk file

What I did was to lift the project files from main folder to the root of project and add a include folder which contains headers which must be seen by the rest of the project: `dns_server.h` for `dns_server()`,
`http_server.h` for `http_server()`  as well as `wifi_manager.h` for `wifi_manager()`, `wifi_manager_set_on_connected()` and `wifi_manager_set_on_disconnected()`. 

Case in point about keeping my component updated, I'd need to use `wifi_manager_set_callback()`

Once the component was done, I added it as a git submodule under [my project](https://github.com/ojousima/ojousima.esp32_blegw.c/tree/master/components) and started the tasks in [main.c](https://github.com/ojousima/ojousima.esp32_blegw.c/blob/master/main/main.c).

JoaoLopesF, 2019-05-21:
HI @tonyp7 and @ojousima, complementing @ojousima explanation:
- Esp-Idf is one way to isolate components (ike lib codes) from main files
In Esp-IDF build system it is made just put our code in folder at components, and create own compoment.mk file. for example for my u8g2 installation, have in component.mk in u8g2 folder:
COMPONENT_SRCDIRS:=csrc
COMPONENT_ADD_INCLUDEDIRS:=csrc
This informe to build system how consider to compile this compoment.
This is clear now ?
About your another considerations, its ok to shutdown the softAp 
Regards

aapris, 2019-07-30:
> * I am planning to implement a ~30s timer once the wifi is connected that will automatically shutdown the AP. It was supposed to be part of release 2.0 but didn't make the cut.

This would be very useful feature. Do you have any estimate when it could be ready?

trombik, 2019-08-25:
#50 has been merged and the library can be imported as an esp-idf component.

tonyp7, 2020-07-28:
Hello,

I consider this closed:

* As of 3.x it's now a component
* Password confirmation: will not be implemented> I do not want to fiddle with the http client side anymore but would accept a PR
* As of 3.1 the access point will automatically close when a connection is established. By default 60s, but it can be configured

closed 2020-07-28 by tonyp7