tonyp7/esp32-wifi-manager #46: cannot build with blank password

mentaluproar, 2019-07-01:
When I try to build, make sets defaults.  menuconfig lets me change those defaults, so I can change the name and password, but there is no option to change auth type.  So when I leave a blank password, the esp just keeps crashing.  Changes in the header dont work since they are overridden by kconfig during the build.

I need a way to set the password blank and the network type to open.

cgfoed, 2019-09-27:
It's not a bug...

The AP starts up with WPA2 in mind - it needs an at least 8 char PW set.
You would need to reconfigure the AP Startup as an unsecure/open station.

I maybe look into that, if I see the need for doing so for my own project.

cgfoed, 2019-10-07:
`	/* SoftAP - Wifi Access Point configuration setup */
	tcpip_adapter_ip_info_t info;
	memset(&info, 0x00, sizeof(info));
	wifi_config_t ap_config = {
		.ap = {
			.ssid_len = 0,
			.channel = wifi_settings.ap_channel,
			.authmode = WIFI_AUTH_OPEN,
			.ssid_hidden = wifi_settings.ap_ssid_hidden,
			.max_connection = DEFAULT_AP_MAX_CONNECTIONS,
			.beacon_interval = DEFAULT_AP_BEACON_INTERVAL,
		},
	};
	memcpy(ap_config.ap.ssid, wifi_settings.ap_ssid , sizeof(wifi_settings.ap_ssid));
	//memcpy(ap_config.ap.password, wifi_settings.ap_pwd, sizeof(wifi_settings.ap_pwd));`

change the authmode and skip the password-memcpy.
Should be as easy as that.

commit 084c24ce (2020-08-01): Merge pull request #98 from tonyp7/big-fixes

fix #46
commit d011124d (2020-08-01): fix #46
closed 2020-08-01 by tonyp7