tonyp7/esp32-wifi-manager #47: Losing wifi credentials

ginodecock, 2019-07-09:
Hi,

I've integrated wifi manager into a battery operated mailbox detection solution. Here I've used the ESP-Wrover module because it can operate on lower voltages. It is powered by 2 x AA.

I notice from time to time that the wifi credentials are lost and needed to be re-configured with the wifi manager ui.

Any idea what can cause this behavior?

Thanks,

ojousima, 2019-07-09:
I've noticed similar behavior on older version of wifi-manager, but I've never been able to figure out how to repeat the issue.

ginodecock, 2019-11-01:
With recent versions I also have the same problem.

kongon, 2020-02-08:
I have also the same problem at latest version.

benjamintyler, 2020-05-26:
I have noticed this as well...sometimes it randomly will need to be reconfigured.

benjamintyler, 2020-06-03:
I think I figured this out. Comment out line 945 and it will fix it.

The sporadic erasing of credentials happens if you lose wifi signal too many times, or the connection times out.

https://github.com/tonyp7/esp32-wifi-manager/blob/master/src/wifi_manager.c#L945

tonyp7, 2020-07-28:
This is fixed as of v3.1

closed 2020-07-28 by tonyp7