tonyp7/esp32-wifi-manager #48: STA Mode server does not host my webpage

AdityaGawali, 2019-07-14:
I have included the wifi manager in my code and what my code does is it hosts a webpage but when I try to open the webpage in my browser it shows your webpage instead of mine even though I am on a home network where esp should be connected as station hosting the webpage.

Also, how can I retrieve the ssid and password from your code so  I can manually start my station code thus solving the above-mentioned problem

tonyp7, 2019-07-15:
You cannot run two different web servers on the same port. You have two choices here:
- Change http_server.c to serve your pages
- Run http_server on a different port than your own http server

AdityaGawali, 2019-07-15:
I change my port to 81 and it works also before closing the issue can you tell where exactly in code are you assigning a static IP? I changed the bool value to 1 but can't find which and where the IP is.
Also a huge Thankyou your project helped me a lot

tonyp7, 2019-07-15:
IP of the server is set by

`/** @brief Defines the access point's default IP address. Default: "10.10.0.1 */
#define DEFAULT_AP_IP						CONFIG_DEFAULT_AP_IP`

You can run make menuconfig to change it or hardcode a value

AdityaGawali, 2019-07-15:
Isn't that the IP of ESP32 as access point, I want to static the IP when it connects to my home router.

tonyp7, 2019-07-16:
This is not possible today, simply because I had no use for it and I was a bit lazy to code the UI for static ip... That being said, you can assign a static IP to the esp32 on your router level through the mac address.

AdityaGawali, 2019-07-16:
No worries Sir, I have made changes to the code for assigning static IP to my home router.

closed 2019-07-16 by AdityaGawali