tonyp7/esp32-wifi-manager #51: Request to Create the wifi_manager_event_group On wifi_manager_start()

qcarver, 2019-08-22:
Hello,

Thank you for your useful contribution.

In my application I do not use the callback feature, instead I read bits from the wifi_manager_event_group.

There is a race condition where if I read from the group before wifi_manager's task is started I will crash trying to read a group that hasn't been created yet.

Why not just create it in the very first method, the start method? 

`
void wifi_manager_start(){

	/* disable the default wifi logging */
	esp_log_level_set("wifi", ESP_LOG_NONE);

	//move to here
	wifi_manager_event_group = xEventGroupCreate();
`

Also, recommend that you initialize it to zero at the top of the file :)

Thanks Again,
QC

tonyp7, 2019-08-25:
Good catch. You are right all objects should be created in wifi_manager_start to avoid threading related issues later on. Flagging as bug for fixing later!

tonyp7, 2019-09-02:
fixed in branch
https://github.com/tonyp7/esp32-wifi-manager/tree/issue%2351

haven't tested it yet but should be fairly straightforward. Will merge after testing.

qcarver, 2019-09-02:
Okay, thanks for the heads up.
V/R,
QC

tonyp7, 2019-10-11:
https://github.com/tonyp7/esp32-wifi-manager/pull/59 fixes the issue -- closing

closed 2019-10-11 by tonyp7