tonyp7/esp32-wifi-manager #52: recording problem

marocama, 2019-08-30:
Hello friends,
I'm starting to learn how to use IDF.
I am trying to write the default_demo example but it is giving the error:
"Executing" ninja all "...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable wifi_manager.elf
FAILED: wifi_manager.elf "

I believe I am installing the wifi_manager library the wrong way.
I'm putting the "default_demo" folder and the "scr" folder in the esp_idf root, is that it?

thank you for now

tonyp7, 2019-09-02:
I cannot comment about your build system. Using standard make all you have to do from the repository is:

```
cd examples/default_demo
make menuconfig
make -j8
```

Where the 8 in -j8 is the number of logical processors you PC has. j8 works with my i7700k.

marocama, 2019-09-02:
Unfortunately I have this return:

"makefile: 10: /make/project.mk: No such file or directory
make: *** No rule to make target `/make/project.mk '. Stop. "

Is this related to the IDF version?

tonyp7, 2019-09-03:
Most likely your environment is not setup correctly. Try to build any esp-idf example, for instance try "make" in esp-idf/examples/get-started/hello_world

marocama, 2019-09-09:
Hello again,
I reinstalled the IDF, and now the project gets to build, but at the end of the build I get this message:

> esptool.py v2.8-dev
> [825/826] Linking CXX executable wifi_manager.elf
> FAILED: wifi_manager.elf

closed 2020-07-03 by tonyp7