tonyp7/esp32-wifi-manager #56: Unable to get wifi list on android Phone.

ankur2854, 2019-09-27:
I am able to connect to ESP32 AP. it open up a HTML page but there is no Wi-Fi list in that page.
I am trying this on android phone as well as on windows phone. Not working on android but the same is working on windows phone.
Can anyone help me out. Thanks in advance.

hanhdt, 2019-09-27:
The library used to Ajax with jQuery v3.2.1 and get WiFi list data from HTTP API. It might be not worked on Android WebView. That is why you don't see anything on Android but it's working fine on other regular browsers (Chrome/Firefox).

ankur2854, 2019-09-27:
I tried with Chrome and Firefox browser on android phone. Still not working.

cgfoed, 2019-09-30:
Did you compile it from scratch? What Android are you using? 
I had this error once when I manipulated the event loop.

be sure that your init
`ESP_ERROR_CHECK(esp_event_loop_init(wifi_manager_event_handler, NULL));`

as well as your event handler
`esp_err_t wifi_manager_event_handler(void *ctx, system_event_t *event) { .... }`
 work properly

ankur2854, 2019-10-01:
I am using Mi Phone (android 8). But i tried with different phones. I am not able to get the list on any android as well as IOS phone. 
On windows laptop it only show list on internet explorer rest no list on other browsers.
Yes, 
ESP_ERROR_CHECK(esp_event_loop_init(wifi_manager_event_handler, NULL)); and wifi_manager_event_handler both are working fine.
[log.txt](https://github.com/tonyp7/esp32-wifi-manager/files/3676264/log.txt)

As I am from embedded background, I have no clue about this issue. Please help me out. 
Thanks in advance.

ankur2854, 2019-10-01:
Please suggest me how to resolve jquery issue.

ankur2854, 2019-10-08:
I got the issue. we got an extra character on server in code.js file due to which browser is unable to render from code.js file.
Error: In http_server.c
Line 186 : netconn_write(conn, code1_js_start, **code1_js_end - code1_js_start -1** , NETCONN_NOCOPY);

tonyp7, 2019-10-08:
@ankur2854 what is your IDE/build system? This issue appears in platform.io but not on a classic make environment as far as I know

closed 2019-10-12 by tonyp7