tonyp7/esp32-wifi-manager #57: no captive portal on Samsung devices

cgfoed, 2019-10-07:
While testing my current project with different devices I noticed that there is no page splash on a samsung device.

Some digging turned out, that they behave a bit special. As solution it is suggested to answer with a vaild 204 packet to trick the android device into thinking there is an internet connection - which makes no sense to me.

Anyhow - I'm trying to implement some code to catch that problem. But any help would be highly appreciated. 
I'll update this Issue as I make any progress.

readings:
https://github.com/tripflex/wifi-captive-portal/issues/7

https://community.arubanetworks.com/t5/Wireless-Access/Samsung-Captive-Portal-Detection/m-p/405934#M78972

https://lemariva.com/blog/2017/11/white-hacking-wemos-captive-portal-using-micropython

https://github.com/tripflex/captive-portal/blob/master/src/mgos_captive_portal.c
starting line 367

https://socifi-doc.atlassian.net/wiki/spaces/SC/pages/94371841/DNS+Fix+to+keep+Android+Splash+Page+and+the+Captive+Portal+Notification+active

tonyp7, 2019-10-12:
I am testing with a Samsung Galaxy Tab A and an iPhone XS and have no issue with the Galaxy Tab.

If all it takes is registering a few end points then sure I can try doing this. Thanks for the tip from the triplex repo

tonyp7, 2019-10-13:
This is what happens on my samsung device when I connect to the esp:

[image]

When I tap on "sign in to the network" I am lead to the captive portal as expected:

[image]

tonyp7, 2019-10-21:
@cgfoed any update? I'll close this issue otherwise

cgfoed, 2019-10-21:
my tests so far:

Honor with Android 9 - worked
iPhone XR - worked
iPhone SE - worked
Galaxy S7 with Android 8 - no captive portal
Galaxy S10 with  Android 9 - no captive portal
iPhone XS - worked 
ZTE #generic_model with Android 8 - worked

hopefully I can test with an Samsung A40 soon. 

Wasn't able to tackle that issue in software so far.
I created a QR-code which links to 10.10.0.1 to slap on as a sticker as workaround - which I guess is useful anyway.

cgfoed, 2019-10-23:
no splash screen on Samsung A40.
It connect's and says there is no internet connection - thats it.

tonyp7, 2019-10-25:
Damn it it really sucks that the only Samsung device I have actually works. Not sure what to do to test it.  I can change the code to add a 200 OK to http://connectivitycheck.gstatic.com/generate_204; which from the forum links you provided seems to trigger the captive portal.

cgfoed, 2019-10-25:
i'll try any code changes, that you can provide.
I also was briefly thinking about sending you a used device, but sg is quite far away ^^'

cgfoed, 2019-11-18:
I don't know why and I don't know how.
But after much trying and further reading I could solve my problem.

The answer for me was this:
https://github.com/esp8266/Arduino/pull/5529/commits/d5265aa5ac4995b65bce3e29025307cfd7ae0d0f#diff-9978b19703e34e9504dcdda781907935

A simple change to
`
/** @brief Defines the access point's default IP address. Default: "10.10.0.1 */
#define DEFAULT_AP_IP						"172.217.28.1"
`
solved all my issues with all Galaxy Devices I've tested so far.

cgfoed, 2019-11-18:
(Maybe this is worth pointing out in the manual somewhere....)

closed 2019-11-18 by cgfoed

cgfoed, 2019-11-18:
F**k! The behavior on iPhone and other Andrioid devices is now horrible. 
So changing the default AP IP is not the solution for my problem here. But maybe I'll figure something out now.

baldhead69, 2020-09-01:
In my samsung edge s7 no page "splash" too.

I install de lib today(01/09/2020).

baldhead69, 2020-09-27:
This issue should be open.

kaly7, 2020-11-21:
It work with Android 10 (Huawei)  , but android 9 not (LG K20).. any ide, why? :(
No captive portal.. if I use http://10.10.0.1 (default ip address) only wait, wait, wait... 
But have any data traffic:

I (34550) dns_server: Replying to DNS request for www.google.com from 10.10.0.2
I (37410) dns_server: Replying to DNS request for alt7-mtalk.google.com from 10.10.0.2
I (37420) dns_server: Replying to DNS request for connectivitycheck.gstatic.com from 10.10.0.2 

:(

Qu35t64, 2021-02-18:
i use IP like 200.200.200.1 and its functional for mi in Samsung devices. I tested in Note8 Android 8.1

 i use this configuration and work for me. https://github.com/Qu35t64/CaptivePortal

cgfoed, 2021-02-22:
> i use IP like 200.200.200.1 and its functional for mi in Samsung devices. I tested in Note8 Android 8.1
> 
> i use this configuration and work for me. https://github.com/Qu35t64/CaptivePortal

But does it still work for iOS and other Android devices?

Qu35t64, 2021-02-23:
> > i use IP like 200.200.200.1 and its functional for mi in Samsung devices. I tested in Note8 Android 8.1
> > i use this configuration and work for me. https://github.com/Qu35t64/CaptivePortal
> 
> But does it still work for iOS and other Android devices?

yes, its work

Inventor-X, 2022-01-08:
Hello Guys...
I have made a captive portal using apache2 server....
I am not able to see the spash screen on samsung devices..only.
I have tried everything that you all have discussed....
But no use.....
Can anyone please guide me what should i do...to get it working on samsung devices...
PLEASE.....

Inventor-X, 2022-01-08:
> > > i use IP like 200.200.200.1 and its functional for mi in Samsung devices. I tested in Note8 Android 8.1
> > > i use this configuration and work for me. https://github.com/Qu35t64/CaptivePortal
> > 
> > 
> > But does it still work for iOS and other Android devices?
> 
> yes, its work

It is not working.....
Can you please suggest anything else......

italocjs, 2025-11-22:
> Still no solutions?

I never figured it out 😕

bsoroudi, 2026-03-13:
This is insane. I have been trying to figure this out for 2 weeks. The solution is to use a public ip range for your hotspot network and captive portal. 

I used the 200.200.200.0/24 subnet. Works great now with samsung, iphone, motorola.