tonyp7/esp32-wifi-manager #58: svg-based icons

cgfoed, 2019-10-11:
I replaced the WiFi-icons with this icons from Alexander Kahlkopf, as I find an vector based images much more pleasing :)

https://iconmonstr.com/lock-26-svg/
https://iconmonstr.com/wireless-1-svg/
https://iconmonstr.com/wireless-2-svg/
https://iconmonstr.com/wireless-3-svg/
https://iconmonstr.com/wireless-4-svg/

check the licence here:
https://iconmonstr.com/license/

compiled code for copy & paste:
```C
.w0 {
    background: url("data:image/svg+xml,%3Csvg xmlns='http://www.w3.org/2000/svg' width='24' height='24' viewBox='0 0 24 24'%3E%3Cpath d='M8.213 16.984c.97-1.028 2.308-1.664 3.787-1.664s2.817.636 3.787 1.664l-3.787 4.016-3.787-4.016zm-1.747-1.854c1.417-1.502 3.373-2.431 5.534-2.431s4.118.929 5.534 2.431l2.33-2.472c-2.012-2.134-4.793-3.454-7.864-3.454s-5.852 1.32-7.864 3.455l2.33 2.471zm-4.078-4.325c2.46-2.609 5.859-4.222 9.612-4.222s7.152 1.613 9.612 4.222l2.388-2.533c-3.071-3.257-7.313-5.272-12-5.272s-8.929 2.015-12 5.272l2.388 2.533z'/%3E%3C/svg%3E") no-repeat right top;
    height: 24px;
    margin-right: 20px;
}
.w1 {
    background: url("data:image/svg+xml,%3Csvg xmlns='http://www.w3.org/2000/svg' width='24' height='24' viewBox='0 0 24 24'%3E%3Cpath d='M8.213 16.984c.97-1.028 2.308-1.664 3.787-1.664s2.817.636 3.787 1.664l-3.787 4.016-3.787-4.016zm-1.747-1.854c1.417-1.502 3.373-2.431 5.534-2.431s4.118.929 5.534 2.431l2.33-2.472c-2.012-2.134-4.793-3.454-7.864-3.454s-5.852 1.32-7.864 3.455l2.33 2.471zm5.534-11.13c3.949 0 7.681 1.517 10.607 4.291l-1.021 1.083c-2.656-2.452-6.023-3.791-9.586-3.791s-6.93 1.339-9.586 3.791l-1.021-1.083c2.926-2.774 6.658-4.291 10.607-4.291zm0-1c-4.687 0-8.929 2.015-12 5.272l2.388 2.533c2.46-2.609 5.859-4.222 9.612-4.222s7.152 1.613 9.612 4.222l2.388-2.533c-3.071-3.257-7.313-5.272-12-5.272z'/%3E%3C/svg%3E") no-repeat right top;
    height: 24px;
    margin-right: 20px;
}
.w2 {
    background: url("data:image/svg+xml,%3Csvg xmlns='http://www.w3.org/2000/svg' width='24' height='24' viewBox='0 0 24 24'%3E%3Cpath d='M8.213 16.984c.97-1.028 2.308-1.664 3.787-1.664s2.817.636 3.787 1.664l-3.787 4.016-3.787-4.016zm3.787-6.78c2.387 0 4.648.876 6.461 2.485l-.969 1.028c-1.556-1.308-3.472-2.018-5.492-2.018-2.021 0-3.937.71-5.492 2.018l-.969-1.028c1.813-1.609 4.075-2.485 6.461-2.485zm0-1c-3.071 0-5.852 1.32-7.864 3.455l2.33 2.472c1.417-1.502 3.373-2.431 5.534-2.431s4.117.929 5.534 2.431l2.33-2.472c-2.012-2.135-4.793-3.455-7.864-3.455zm0-5.204c3.949 0 7.682 1.517 10.607 4.291l-1.021 1.083c-2.656-2.452-6.023-3.791-9.586-3.791s-6.93 1.339-9.586 3.791l-1.021-1.083c2.926-2.774 6.658-4.291 10.607-4.291zm0-1c-4.687 0-8.929 2.015-12 5.272l2.388 2.533c2.46-2.609 5.859-4.222 9.612-4.222 3.754 0 7.152 1.613 9.611 4.222l2.389-2.533c-3.071-3.257-7.313-5.272-12-5.272z'/%3E%3C/svg%3E") no-repeat right top;
    height: 24px;
    margin-right: 20px;
}
.w3 {
    background: url("data:image/svg+xml,%3Csvg xmlns='http://www.w3.org/2000/svg' width='24' height='24' viewBox='0 0 24 24'%3E%3Cpath d='M12 16.32c.843 0 1.65.257 2.345.735l-2.345 2.487-2.345-2.487c.695-.478 1.502-.735 2.345-.735zm0-1c-1.479 0-2.817.636-3.787 1.664l3.787 4.016 3.787-4.016c-.97-1.028-2.308-1.664-3.787-1.664zm0-5.116c2.387 0 4.648.876 6.461 2.485l-.969 1.028c-1.556-1.308-3.472-2.018-5.492-2.018-2.021 0-3.937.71-5.492 2.018l-.969-1.028c1.813-1.609 4.074-2.485 6.461-2.485zm0-1c-3.071 0-5.852 1.32-7.864 3.455l2.33 2.472c1.417-1.502 3.373-2.431 5.534-2.431s4.117.929 5.534 2.431l2.33-2.472c-2.012-2.135-4.793-3.455-7.864-3.455zm0-5.204c3.949 0 7.682 1.517 10.607 4.291l-1.021 1.083c-2.656-2.452-6.023-3.791-9.586-3.791s-6.93 1.339-9.586 3.791l-1.021-1.083c2.925-2.774 6.658-4.291 10.607-4.291zm0-1c-4.687 0-8.929 2.015-12 5.272l2.389 2.533c2.459-2.609 5.857-4.222 9.611-4.222s7.152 1.613 9.611 4.222l2.389-2.533c-3.071-3.257-7.313-5.272-12-5.272z'/%3E%3C/svg%3E") no-repeat right top;
    height: 24px;
    margin-right: 20px;
}
.pw {
    background: url("data:image/svg+xml,%3Csvg xmlns='http://www.w3.org/2000/svg' width='24' height='24' viewBox='0 0 24 24'%3E%3Cpath d='M17 9.761v-4.761c0-2.761-2.238-5-5-5-2.763 0-5 2.239-5 5v4.761c-1.827 1.466-3 3.714-3 6.239 0 4.418 3.582 8 8 8s8-3.582 8-8c0-2.525-1.173-4.773-3-6.239zm-8-4.761c0-1.654 1.346-3 3-3s3 1.346 3 3v3.587c-.927-.376-1.938-.587-3-.587s-2.073.211-3 .587v-3.587zm3 17c-3.309 0-6-2.691-6-6s2.691-6 6-6 6 2.691 6 6-2.691 6-6 6zm2-6c0 1.104-.896 2-2 2s-2-.896-2-2 .896-2 2-2 2 .896 2 2z'/%3E%3C/svg%3E") no-repeat right top;
    height: 24px;
    margin-right: 20px;
    height: 24px;
    margin-right: 30px;
}
```

[image]

tonyp7, 2019-10-11:
Love it! Although embedding this into a MIT licensed code might an issue because of this clause:

"Licensee may not sell, redistribute, **_sublicense_**, share or otherwise transfer the Work to other people or entities." 

I'll contact the author and see what he has to say.

tonyp7, 2019-10-14:
FWIW:

> Fri, Oct 11, 8:11 PM
> 
> Hello Alexander,
> 
> I maintain an embedded application which is open-sourced under MIT.
> 
> See:
> https://github.com/tonyp7/esp32-wifi-manager
> 
> One user suggested using your icons:
> https://github.com/tonyp7/esp32-wifi-manager/issues/58
> 
> And I agree they are awesome. However, since you are using a proprietary mostly copy-left license for your work, I am unsure whether or not this works.
> 
> The clause:
> Licensee may not sell, redistribute, sublicense, share or otherwise transfer the Work to other people or entities.
> 
> Seems to mean I cannot embed your icons into my app since that would effectively mean sub licensing and sharing the work to other people.
> 
> Would you be so kind as to license your work; at least for the images listed under the issue with something without any ambiguity such as CC-BY or MIT?
> 
> Thanks,
> Tony

tonyp7, 2020-07-01:
I never had an answer from Alexander regarding the usage of these icons so I am therefore closing this issue.

It's unfortunate, because they are really nice, but I can't risk compromising the license of this software.

closed 2020-07-01 by tonyp7