tonyp7/esp32-wifi-manager #60: Any Known Issues Accessing Webpage Using PC?

sullij, 2019-10-11:
Hello, I am using the wifi manager for my application and it works great within a captive portal, or on any mobile device that I have tried so far.

But for some reason when connecting to the ESP's SSID using a Windows PC, then accessing the webpage, the page isn't as reliable and sometimes the ESP disconnects from my PC and I have to bring up the ESP's SSID again on my WiFi list. 

Without going into the specific errors coming up in the terminal monitor, I just wanted to ask if this is a known issue, or if the wifi manager isn't meant for use outside of mobile devices? 

many thanks again for your excellent contribution

best!

cgfoed, 2019-10-14:
Is your Windows PC connected additionally to another network (via cable maybe?) ?
I had similar issues with a windows machine which was hooked up to an Ethernet connection. 
When I unplugged it everything worked fine.

sullij, 2019-10-14:
I just tried again (without any changes to the wifi manager) without any other network connection on a Windows PC and occasionally I get an assertion similar to the below:

[image]

I tried with Chrome, Firefox and Explorer and the same occurs.

I am using esp-idf version: v4.0-dev-1346-g1b6010bed
 and toolchain: xtensa-esp32-elf-gcc.exe (crosstool-NG esp32-2019r1) 8.2.0

Any suggestions, or ideas, are much appreciated

tonyp7, 2019-10-15:
Very interesting. I haven't tried the 4.0 as it hasn't officially released yet. I'm on 3.2. I'll try to replicate the behavior. Could you confirm you are using the current master branch version of the wifi manager?

tonyp7, 2019-10-15:
I think I have found the issue. In case netconn_accept doesn't return ERR_OK (most likely ERR_TIMEOUT) the server crashes and burn in a dumpster fire. Why this happens more with a laptop I do not know, but it is however a very simple fix.
I'll get a PR asap!

tonyp7, 2019-10-15:
try the latest commit!

sullij, 2019-10-15:
Thanks so much! 

I tried updating the http_server.c file and I'm still getting an error like this when using Windows PC:

assertion "netconn marked closed" failed: file "C:......./components/lwip/lwip/src/api/api_msg.c", line 860, function: netconn_drain
abort() was called at PC 0x400e6bf7 on core 1
0x400e6bf7: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

But, I will try this again this evening and be sure I am not missing something, many thanks!

tonyp7, 2019-10-16:
Would you mind posting the full dump?

This more and more looks like a bug with esp-idf 4.0 to be honest. I'll try to compile with the current master branch of esp-idf and see if I can reproduce

See: 
https://github.com/espressif/esp-idf/issues/3842
https://github.com/yarrick/lwip/blob/1a10a942f22c5d139573c5a4b68431a57ea47d30/src/api/api_msg.c#L834

sullij, 2019-10-16:
I see... yes, I ended up with similar results yesterday evening using esp-idf 4.0. 

I'll give it another go tonight and post the results and also try using esp-idf version 3.2 as well.

sullij, 2019-10-17:
After trying the updated wifi manager using esp-idf v3.2, I agree, this seems to be an issue with idf version 4.0. Using v3.2 so far I do not see any errors on the monitor and the webpage updates the list of APs without disconnecting 👍  thx very much again and if I catch anything else regarding this, I'll update, but in the meantime I'll switch to v3.2!

tonyp7, 2020-07-03:
Try again with the latest commit, this should fix a lot of things on 4.x

closed 2020-07-03 by tonyp7