tonyp7/esp32-wifi-manager #66: 8KB of task stack?

malinges, 2020-02-14:
Hi,

I'd like to start by thanking you for your work, your library is great and I use it regularly :)
However I'm a bit puzzled by the "8KB of task stack in total" part in the README.
I found 3 tasks in the code, each with its own stack size:

- dns_server: 3072
- http_server: 2048
- wifi_manager: 4096

This adds up to 9216.
However these are [not bytes but words](https://www.freertos.org/a00125.html), which on the ESPs are 4 bytes (32 bits), so this gives a total of 36 864 bytes, or 36KB. Quite more than 8KB :/
Am I missing something?

BrianPugh, 2020-05-08:
esp-idf's FreeRTOS deviates from vanilla FreeRTOS when it comes to their stack size declaration. They give it in bytes:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/freertos-smp.html#tasks-and-task-creation

malinges, 2020-05-11:
@BrianPugh OMG you're right... I was indeed missing something. Thanks!
So we're down to 9KB (instead of 36KB) vs. the documented 8KB. Much better :-)

tonyp7, 2020-07-08:
I have completely revised the readme file to include more documentation and dropped this reference to the stack.

closed 2020-07-08 by tonyp7