tonyp7/esp32-wifi-manager #67: Constantly rebooting ESP32 module

mgoeller, 2020-02-18:
I integrated your wifi manager into my ESP32 doorbell project, I however have stability issues with the ESP32 constantly rebooting (which to my understanding has nothing to do with my App, because it still happens in the previous steps):

It either happens after:
```
I (800) dns_server: DNS Server listening on 53/udp
ets Jun  8 2016 00:22:57
```

or after:
```
I (612) cpu_start: Pro cpu start user code
ets Jun  8 2016 00:22:57
```

Full logs attached:
```
mailgoe@MacBook-Pro-2 doorbell % miniterm.py /dev/tty.usbserial-00000000 115200 --raw
--- Miniterm on /dev/tty.usbserial-00000000  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6928
ho 0 tail 12 room 4
load:0x40078000,len:14724
load:0x40080400,len:4292
entry 0x400806e4
I (75) boot: Chip Revision: 1
I (75) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot: ESP-IDF v4.0-beta2-295-g20e6d46d4-dirty 2nd stage bootloader
I (41) boot: compile time 13:33:09
I (42) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (52) boot: SPI Mode       : DIO
I (56) boot: SPI Flash Size : 4MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x310f4 (200948) map
I (185) esp_image: segment 1: paddr=0x0004111c vaddr=0x3ffb0000 size=0x03710 ( 14096) load
I (190) esp_image: segment 2: paddr=0x00044834 vaddr=0x40080000 size=0x00400 (  1024) load
I (192) esp_image: segment 3: paddr=0x00044c3c vaddr=0x40080400 size=0x0b3d4 ( 46036) load
I (219) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x945dc (607708) map
I (435) esp_image: segment 5: paddr=0x000e45fc vaddr=0x4008b7d4 size=0x09a7c ( 39548) load
I (465) boot: Loaded app from partition at offset 0x10000
I (465) boot: Disabling RNG early entropy source...
I (465) cpu_start: Pro cpu up.
I (469) cpu_start: Application information:
I (474) cpu_start: Project name:     doorlock
I (479) cpu_start: App version:      e08781d-dirty
I (485) cpu_start: Compile time:     Jan 30 2020 13:33:13
I (491) cpu_start: ELF file SHA256:  36f0467429ffbd1e...
I (497) cpu_start: ESP-IDF:          v4.0-beta2-295-g20e6d46d4-dirty
I (504) cpu_start: Starting app cpu, entry point is 0x400811c4
I (0) cpu_start: App cpu up.
I (514) heap_init: Initializing. RAM available for dynamic allocation:
I (521) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (527) heap_init: At 3FFB9F28 len 000260D8 (152 KiB): DRAM
I (534) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (540) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (546) heap_init: At 40095250 len 0000ADB0 (43 KiB): IRAM
I (553) cpu_start: Pro cpu start user code
[... 260 lines cut ...]
I (580) MainProg: 

===== D O O R L O C K   A C C E S S O R Y =====

I (620) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (620) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (630) wifi_manager: Set STA IP String to: 0.0.0.0
I (650) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (650) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (680) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (680) wifi_manager: wifi_manager: Start DHCP client for STA interface. If not already running
I (780) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (780) wifi_manager: MESSAGE: ORDER_LOAD_AND_RESTORE_STA
I (780) wifi_manager: No saved wifi found on startup. Starting access point.
I (780) wifi_manager: SYSTEM_EVENT_STA_START
I (790) wifi_manager: MESSAGE: ORDER_START_AP
I (780) http_server: HTTP Server listening on 80/tcp
I (800) wifi_manager: SYSTEM_EVENT_AP_START
I (800) dns_server: DNS Server listening on 53/udp
I (22120) wifi_manager: SYSTEM_EVENT_AP_STACONNECTED
I (22660) tcpip_adapter: softAP assign IP to station,IP is: 10.10.0.2
I (22970) dns_server: Replying to DNS request for captive.apple.com from 10.10.0.2
I (27760) dns_server: Replying to DNS request for 17-courier.push.apple.com from 10.10.0.2
I (28210) wifi_manager: SYSTEM_EVENT_AP_STADISCONNECTED
```

closed 2020-02-24 by mgoeller

tonyp7, 2020-02-25:
This should be because of an issue with esp-idf 4.0. Try esp-idf 3.x ?

mgoeller, 2020-02-25:
My problem was my hardware, the EN pin was not pulled up to VCC.