tonyp7/esp32-wifi-manager #69: build steps PlatformIo

manju-reddys, 2020-03-21:
Hi,

Is it possible to out the steps to build it in PlatformIo IDE plz?

tonyp7, 2020-07-03:
Sorry I do not use platform.io. The project builds fine using either make or cmake

closed 2020-07-03 by tonyp7

gagank1, 2021-01-31:
#### Fix for platformio/platform-espressif32 v3.0.0 and ESP-IDF v4.2

Download release v3.3.1, unzip and add it to your `components/` folder in the root of your project.

Add this to your platform.ini:
```
board_build.embed_txtfiles = 
  components/esp32-wifi-manager/src/index.html
  components/esp32-wifi-manager/src/style.css
  components/esp32-wifi-manager/src/code.js
```

Then modify these 3 lines in http_app.c:

252: `httpd_resp_send(req, (char*)index_html_start, index_html_end - index_html_start);`
258: `httpd_resp_send(req, (char*)code_js_start, code_js_end - code_js_start);`
265: `httpd_resp_send(req, (char*)style_css_start, style_css_end - style_css_start);`

to
```
httpd_resp_send(req, (char*)index_html_start, strlen((char*)index_html_start));
httpd_resp_send(req, (char*)code_js_start, strlen((char*)code_js_start));
httpd_resp_send(req, (char*)style_css_start, strlen((char*)style_css_start));
```