tonyp7/esp32-wifi-manager #70: iOS 13.4 blank screen

mgoeller, 2020-04-02:
Somehow iOS13.4 and newest macOS release have an error: the wifi manager - if it connects at all - often just returns a blank page and doesn't load the html.

This [repo](https://github.com/RavenSystem/esp-homekit-devices/commit/25a06a76e314a716b5c6d97b855e8817092bd304) is using a esp8266 wifi manager and just fixed this error (or did a workaround since it may also be on apple side), however I do not understand what needs to be changed in esp32-wifi-manager.

Can anyone help?

ginodecock, 2020-04-16:
I can report the same issue with esp32 and ios 13.4.

mgoeller, 2020-04-16:
In another repo this was fixed for ESP8266, but I don't understand what exactly was done. Respectively I don't see the difference in any of the changes to the state before.

See: https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_1.9.2

ginodecock, 2020-05-02:
I switched to ble provisioning.

mgoeller, 2020-05-02:
Can you link me to an article / repo explaining that?

ginodecock, 2020-05-02:
This is an example of Espressif https://github.com/espressif/esp-idf/tree/master/examples/provisioning/legacy/ble_prov

tonyp7, 2020-07-04:
I have upgraded to iOS 13.5 from 13.2 and confirms the captive portal is no longer kicked in

tonyp7, 2020-07-04:
@mgoeller thanks for the pointer. I really don't understand what they changed and I can't find any relevant information about what Apple changed to break the portal.

What I see is that the /GET request does not even run on the server. That means the program does not even get the chance to read the host requested.

I opened an issue on their repo and hopefully they can give some pointers.

tonyp7, 2020-07-05:
#84 fixes this issue

PR #84 (2020-07-05): Esp idf 4 update
Fixes #70

Turns out updating the code to support the latest IP library (esp_netif) fixes the issue.
closed 2020-07-05 by tonyp7