tonyp7/esp32-wifi-manager #71: ESP_ERROR_CHECK fails during user connection

akhilv1, 2020-04-06:
Hi, I found a bug in the connection process where sometimes the event ORDER_START_WIFI_SCAN occurs during the execution of ORDER_CONNECT_STA which causes the esp to reset due to an assert failure for the scan because the sta is attempting to connect. 

I fixed this by adding a boolean that gets set to true in ORDER_CONNECT_STA if msg.param == CONNECTION_REQUEST_USER, and only performing the wifi scan when the flag is false. this flag needs to be reset to false if the password is wrong under EVENT_STA_DISCONNECTED.

Skinner927, 2020-04-25:
What about using `ESP_ERROR_CHECK_WITHOUT_ABORT` instead of `ESP_ERROR_CHECK`?
https://github.com/tonyp7/esp32-wifi-manager/blob/master/src/wifi_manager.c#L776

Or is it advantageous to restart the device?

tonyp7, 2020-07-30:
@akhilv1 a boolean is not an acceptable thread synchronization. However I think this can easily be fixed by checking for the event group bits CONNECTION_REQUEST_USER or CONNECTION_REQUEST_RESTORE_CONNECTION.

I mark this as bug and include in milestone 3.2

commit 43392e6a (2020-08-01): fix #71
commit b45e5ee9 (2020-08-01): Merge pull request #99 from tonyp7/bug-fixes

fix #71
closed 2020-08-01 by tonyp7