tonyp7/esp32-wifi-manager #78: Unreliable performance on esp-idf v4.1

JasonFevang, 2020-06-03:
Three commons errors while using this component with esp-idf v4.1
1. When requesting the webpage, or when sending the esp32 wifi credentials, the error at the bottom occurs occasionally. See #60 
```
assertion "PCB must be deallocated outside this function" failed: file "C:/Users/Jason/esp-idf/components/lwip/lwip/src/api/api_msg.c", line 827, function: netconn_free
abort() was called at PC 0x40130c43 on core 0
0x40130c43: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

ELF file SHA256: 11bfde8b20211129

Backtrace: 0x4008f869:0x3ffc8cb0 0x4008fbed:0x3ffc8cd0 0x40130c43:0x3ffc8cf0 0x400e01d7:0x3ffc8d20 0x400def26:0x3ffc8d40 0x400d6aae:0x3ffc8d60 0x4009219d:0x3ffc8d90
0x4008f869: invoke_abort at C:/Users/Jason/esp-idf/components/esp32/panic.c:155

0x4008fbed: abort at C:/Users/Jason/esp-idf/components/esp32/panic.c:172

0x40130c43: __assert_func at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400e01d7: netconn_free at C:/Users/Jason/esp-idf/components/lwip/lwip/src/api/api_msg.c:827 (discriminator 1)

0x400def26: netconn_delete at C:/Users/Jason/esp-idf/components/lwip/lwip/src/api/api_lib.c:257

0x400d6aae: http_server at C:\Users\Jason\Documents\work\esp-idf\wifi-manager-test\build/../components/esp32-wifi-manager/src/http_server.c:115

0x4009219d: vPortTaskWrapper at C:/Users/Jason/esp-idf/components/freertos/port.c:143

Rebooting...
```
2. When requesting the web page, the web page never loads, or else takes a long time(a minute) to load
3. The web page loads, but the scanned wifi ssid list doesn't load until a manual page refresh is issued

Details:
These errors happen when configuring with phone or PC
Very occasionally configuration is successful, but mostly it doesn't work
Running the provided example only
Running the latest commit 2871cf1fab3fa5c60c67fcee5c0d0e0c395db34c

Workaround Solution:
Downgrading my esp-idf version to v3.3.2 removes these issues and the wifi-manager works without problems so far

tonyp7, 2020-06-03:
Hello Jason,

You are correct, there is an incompatibility with 4.0+ but I have been unable to locate it.

To be fair, this wifi manager needs to be rebuilt from the ground. I'd like to implement a proper Queue based message pump for it but I haven't got the time nor motivation to do so. You're stuck on 3.3 LTS for now but hopefully this shouldn't last.

JasonFevang, 2020-06-04:
That's fair. I'm good with 3.3LTS for now. If decide to rewrite it myself
and make any progress I'll let you know

tonyp7, 2020-07-03:
Should be good on 4.x now!

closed 2020-07-03 by tonyp7