tonyp7/esp32-wifi-manager #89: Manual connection issue

MoustachedBird, 2020-07-15:
When I try to use a Manual Connection (Hidden SSID) it has a problem when I click "JOIN" button nothing happens... Any Idea? 

Thank you :)

MoustachedBird, 2020-07-15:
Sorry, it think I already solved it (i guess...), the problem seemed to be this section in the **code.js** file
```

  gel("cancel").addEventListener("click", cancel, false);

  gel("manual_cancel").addEventListener("click", cancel, false);

  gel("join").addEventListener("click", performConnect, false);

  gel("manual_join").addEventListener(
    "click",
    (e) => {
      performConnect(e.data("connect"));
    },
    false
  );
```
I only changed manual section by 

```
 gel("manual_join").addEventListener(
    "click",
    (e) => {
      performConnect("manual");
    },
    false
  );
```

tonyp7, 2020-07-15:
I fixed a few bugs due to rewriting code.js and ditching jquery. Hopefully this is the last one. Good catch @MoustachedBird

commit 75a6e2ba (2020-07-15): Fix #89
commit a7fbbba1 (2020-07-15): Merge pull request #91 from tonyp7/bug-fixes

Fix #89
closed 2020-07-15 by tonyp7