tonyp7/esp32-wifi-manager #95: Cannot get web browser !

lktinh2018, 2020-07-22:
### Prerequisites

- [x] I have written a descriptive issue title
- [ ] I have verified that I am using the latest version of esp32-wifi-manager
- [ ] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description

Hello author, I use default example and connect with AP but I cannot open web normally.
It's always show 404 error ? 
### Steps to Reproduce

### System Configuration

- Environment (Operating system, version and so on):
- Additional information:

tonyp7, 2020-07-22:
Hello,

Can you navigate to http://10.0.0.1/wifimanager ?

I recently moved the URL so that it’s easier to coexist with your own webapp, but now I realize it makes it less user friendly in this case.

lktinh2018, 2020-07-22:
Hi,
I tried all, default ip 10.10.0.1 and 10.10.0.1/wifimanager.
But now, I can use it normally after swtiching to tag v3.0

tonyp7, 2020-07-22:
Ok. I’ll revert this change and set it as an optional feature for those that need it.

tonyp7, 2020-07-23:
Update to master, it should be fixed now (#93)

closed 2020-07-23 by tonyp7