tonyp7/esp32-wifi-manager #97: idf.py build fails with esp_netif.h: No such file or directory

ifordh, 2020-07-26:
### Prerequisites

- [x] I have written a descriptive issue title
- [x ] I have verified that I am using the latest version of esp32-wifi-manager
- [x ] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description
Output of idf.py build after following the intructions to include in a project. the new project is a copy of hello_world

C:\Dropbox\Projects\wifi-soft-setup>idf.py build
Checking Python dependencies...
Python requirements from C:\DEV\ESP32\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\dropbox\projects\wifi-soft-setup\build
Executing "ninja all"...
[1/13] cmd.exe /C "cd /D C:\Dropbox\Projects\wifi-soft-set...*********************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/13] Performing build step for 'bootloader'
ninja: no work to do.
[3/11] Building C object esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/http_app.c.obj
FAILED: esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/http_app.c.obj
ccache C:\DEV\ESP32\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/esp32-wifi-manager/src -IC:/DEV/ESP32/esp-idf/components/newlib/platform_include -IC:/DEV/ESP32/esp-idf/components/freertos/include -IC:/DEV/ESP32/esp-idf/components/heap/include -IC:/DEV/ESP32/esp-idf/components/log/include -IC:/DEV/ESP32/esp-idf/components/soc/esp32/include -IC:/DEV/ESP32/esp-idf/components/soc/include -IC:/DEV/ESP32/esp-idf/components/esp_rom/include -IC:/DEV/ESP32/esp-idf/components/esp_common/include -IC:/DEV/ESP32/esp-idf/components/xtensa/include -IC:/DEV/ESP32/esp-idf/components/xtensa/esp32/include -IC:/DEV/ESP32/esp-idf/components/esp32/include -IC:/DEV/ESP32/esp-idf/components/driver/include -IC:/DEV/ESP32/esp-idf/components/esp_ringbuf/include -IC:/DEV/ESP32/esp-idf/components/esp_event/include -IC:/DEV/ESP32/esp-idf/components/tcpip_adapter/include -IC:/DEV/ESP32/esp-idf/components/lwip/include/apps -IC:/DEV/ESP32/esp-idf/components/lwip/include/apps/sntp -IC:/DEV/ESP32/esp-idf/components/lwip/lwip/src/include -IC:/DEV/ESP32/esp-idf/components/lwip/port/esp32/include -IC:/DEV/ESP32/esp-idf/components/lwip/port/esp32/include/arch -IC:/DEV/ESP32/esp-idf/components/vfs/include -IC:/DEV/ESP32/esp-idf/components/esp_wifi/include -IC:/DEV/ESP32/esp-idf/components/esp_wifi/esp32/include -IC:/DEV/ESP32/esp-idf/components/esp_eth/include -IC:/DEV/ESP32/esp-idf/components/efuse/include -IC:/DEV/ESP32/esp-idf/components/efuse/esp32/include -IC:/DEV/ESP32/esp-idf/components/app_trace/include -IC:/DEV/ESP32/esp-idf/components/nvs_flash/include -IC:/DEV/ESP32/esp-idf/components/spi_flash/include -IC:/DEV/ESP32/esp-idf/components/mbedtls/port/include -IC:/DEV/ESP32/esp-idf/components/mbedtls/mbedtls/include -IC:/DEV/ESP32/esp-idf/components/mdns/include -IC:/DEV/ESP32/esp-idf/components/console -IC:/DEV/ESP32/esp-idf/components/wpa_supplicant/include -IC:/DEV/ESP32/esp-idf/components/wpa_supplicant/port/include -IC:/DEV/ESP32/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/DEV/ESP32/esp-idf/components/esp_http_server/include -IC:/DEV/ESP32/esp-idf/components/nghttp/port/include -IC:/DEV/ESP32/esp-idf/components/nghttp/nghttp2/lib/includes -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/http_app.c.obj -MF esp-idf\esp32-wifi-manager\CMakeFiles\__idf_esp32-wifi-manager.dir\src\http_app.c.obj.d -o esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/http_app.c.obj   -c ../components/esp32-wifi-manager/src/http_app.c
../components/esp32-wifi-manager/src/http_app.c:43:10: fatal error: esp_netif.h: No such file or directory
 #include "esp_netif.h"
          ^~~~~~~~~~~~~
compilation terminated.
[4/11] Building C object esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/wifi_manager.c.obj
FAILED: esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/wifi_manager.c.obj
ccache C:\DEV\ESP32\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/esp32-wifi-manager/src -IC:/DEV/ESP32/esp-idf/components/newlib/platform_include -IC:/DEV/ESP32/esp-idf/components/freertos/include -IC:/DEV/ESP32/esp-idf/components/heap/include -IC:/DEV/ESP32/esp-idf/components/log/include -IC:/DEV/ESP32/esp-idf/components/soc/esp32/include -IC:/DEV/ESP32/esp-idf/components/soc/include -IC:/DEV/ESP32/esp-idf/components/esp_rom/include -IC:/DEV/ESP32/esp-idf/components/esp_common/include -IC:/DEV/ESP32/esp-idf/components/xtensa/include -IC:/DEV/ESP32/esp-idf/components/xtensa/esp32/include -IC:/DEV/ESP32/esp-idf/components/esp32/include -IC:/DEV/ESP32/esp-idf/components/driver/include -IC:/DEV/ESP32/esp-idf/components/esp_ringbuf/include -IC:/DEV/ESP32/esp-idf/components/esp_event/include -IC:/DEV/ESP32/esp-idf/components/tcpip_adapter/include -IC:/DEV/ESP32/esp-idf/components/lwip/include/apps -IC:/DEV/ESP32/esp-idf/components/lwip/include/apps/sntp -IC:/DEV/ESP32/esp-idf/components/lwip/lwip/src/include -IC:/DEV/ESP32/esp-idf/components/lwip/port/esp32/include -IC:/DEV/ESP32/esp-idf/components/lwip/port/esp32/include/arch -IC:/DEV/ESP32/esp-idf/components/vfs/include -IC:/DEV/ESP32/esp-idf/components/esp_wifi/include -IC:/DEV/ESP32/esp-idf/components/esp_wifi/esp32/include -IC:/DEV/ESP32/esp-idf/components/esp_eth/include -IC:/DEV/ESP32/esp-idf/components/efuse/include -IC:/DEV/ESP32/esp-idf/components/efuse/esp32/include -IC:/DEV/ESP32/esp-idf/components/app_trace/include -IC:/DEV/ESP32/esp-idf/components/nvs_flash/include -IC:/DEV/ESP32/esp-idf/components/spi_flash/include -IC:/DEV/ESP32/esp-idf/components/mbedtls/port/include -IC:/DEV/ESP32/esp-idf/components/mbedtls/mbedtls/include -IC:/DEV/ESP32/esp-idf/components/mdns/include -IC:/DEV/ESP32/esp-idf/components/console -IC:/DEV/ESP32/esp-idf/components/wpa_supplicant/include -IC:/DEV/ESP32/esp-idf/components/wpa_supplicant/port/include -IC:/DEV/ESP32/esp-idf/components/wpa_supplicant/include/esp_supplicant -IC:/DEV/ESP32/esp-idf/components/esp_http_server/include -IC:/DEV/ESP32/esp-idf/components/nghttp/port/include -IC:/DEV/ESP32/esp-idf/components/nghttp/nghttp2/lib/includes -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0.1-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/wifi_manager.c.obj -MF esp-idf\esp32-wifi-manager\CMakeFiles\__idf_esp32-wifi-manager.dir\src\wifi_manager.c.obj.d -o esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/wifi_manager.c.obj   -c ../components/esp32-wifi-manager/src/wifi_manager.c
../components/esp32-wifi-manager/src/wifi_manager.c:46:10: fatal error: esp_netif.h: No such file or directory
 #include "esp_netif.h"
          ^~~~~~~~~~~~~
compilation terminated.
[5/11] Building C object esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/dns_server.c.obj
FAILED: esp-idf/esp32-wifi-manager/CMakeFiles/__idf_esp32-wifi-manager.dir/src/dns_server.c.obj
[... cut ...]
../components/esp32-wifi-manager/src/dns_server.c:100:28: error: request for member 'ip' in something not a structure or union
     ra.sin_addr.s_addr = ip.ip.addr;
                            ^
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Dropbox\Projects\wifi-soft-setup>

### Steps to Reproduce
Follow steps to include in your project and then idf.py build

### System Configuration
hello_world example

- Environment (Operating system, version and so on):
stable esp-idf.
Windows 10

tonyp7, 2020-07-27:
Hello,

I will just refer you to the requirements in the README page:

# Getting Started

## Requirements

To get you started, esp32-wifi-manager needs:

- esp-idf 4.1 and up
- esp32 or esp32-s2

Due to breaking changes in esp-idf 4.1, most notably the complete revision on how the event wifi loop works and how the tcpip library was deprecated in favour of a newer library (esp_netif), esp32-wifi-manager will not work with older releases of espressif's frameworks

pablacho, 2020-07-27:
Hello, I am having the exact same issue. I have esp-idf latest build (4.0.1) and the same file

tonyp7, 2020-07-27:
Yes and 4.0.1 clearly does not match the requirement on the esp-idf version, which is listed as

# esp-idf 4.1 and up

closed 2020-07-27 by tonyp7