tonyp7/esp32-wifi-manager #101: DEFAULT_STA_ONLY do not change

Bertoleti, 2020-08-05:
Hi.

I think I found a bug on DEFAULT_STA_ONLY configuration.

If I change the value (default: 1) to 0, the value never change on saved settings.

If I want to change this value, I need to force a value before compare with a saved value:

`
wifi_settings.sta_only = DEFAULT_STA_ONLY;

		if( (esp_err == ESP_OK  || esp_err == ESP_ERR_NVS_NOT_FOUND) &&
				(
				strcmp( (char*)tmp_settings.ap_ssid, (char*)wifi_settings.ap_ssid) != 0 ||
				strcmp( (char*)tmp_settings.ap_pwd, (char*)wifi_settings.ap_pwd) != 0 ||
				tmp_settings.ap_ssid_hidden != wifi_settings.ap_ssid_hidden ||
				tmp_settings.ap_bandwidth != wifi_settings.ap_bandwidth ||
				tmp_settings.sta_only != wifi_settings.sta_only ||
				tmp_settings.sta_power_save != wifi_settings.sta_power_save ||
				tmp_settings.ap_channel != wifi_settings.ap_channel
				)
		){
			esp_err = nvs_set_blob(handle, "settings", &wifi_settings, sizeof(wifi_settings));
			if (esp_err != ESP_OK) return esp_err;
			change = true;

			ESP_LOGD(TAG, "wifi_manager_wrote wifi_settings: SoftAP_ssid: %s",wifi_settings.ap_ssid);
			ESP_LOGD(TAG, "wifi_manager_wrote wifi_settings: SoftAP_pwd: %s",wifi_settings.ap_pwd);
			ESP_LOGD(TAG, "wifi_manager_wrote wifi_settings: SoftAP_channel: %i",wifi_settings.ap_channel);
			ESP_LOGD(TAG, "wifi_manager_wrote wifi_settings: SoftAP_hidden (1 = yes): %i",wifi_settings.ap_ssid_hidden);
			ESP_LOGD(TAG, "wifi_manager_wrote wifi_settings: SoftAP_bandwidth (1 = 20MHz, 2 = 40MHz): %i",wifi_settings.ap_bandwidth);
			ESP_LOGD(TAG, "wifi_manager_wrote wifi_settings: sta_only (0 = APSTA, 1 = STA when connected): %i",wifi_settings.sta_only);
			ESP_LOGD(TAG, "wifi_manager_wrote wifi_settings: sta_power_save (1 = yes): %i",wifi_settings.sta_power_save);
		}`

tonyp7, 2020-08-06:
This just needs to be cleaned up. wifi_settings.sta_only serves absolutely no purpose. In the latest version the wifi manager will connect to a saved wifi on boot without kickstarting the AP.

Also, the AP shutdowns automatically after 60s upon a successful connection.

In effect this setting should be removed as it is now completely useless.

What are you trying to achieve?

Bertoleti, 2020-08-06:
Hmm... I see.
Actualy, what I woul like to do is change the SSID and password "on the flight".
Let's supose that I have connected on a network and I want to change it. If I have the STA mode always on, I just need to connect and change the parameters.
Or I thinking about a way to erase the saved parameters using a reset button on a GPIO.

I think that if I want to change these parameters now, I need to turn off the router, so the STA mode will start and then I can change, right?

Is there another way to do that?

tonyp7, 2020-08-06:
Once you’re connected to a wifi, just connect your iPhone or laptop or anything else on the same network and use the IP address of the esp32. That way you can access the web config as well and disconnect the network if you need to.

Bertoleti, 2020-08-06:
Yes. But the IP address given by a DHCP can change, or, the user probably won't know what IP was given to device, right?

Imagine a user that put the device on network and don't have idea of witch IP was given to it.
If the user would like to change, the way that I see now is reset device with a button pressed (for instance) erase the parameters and start the library.

I don't know if I have other way to do that.

Thanks in advance

tonyp7, 2020-08-07:
Just comment out this entire section:

https://github.com/tonyp7/esp32-wifi-manager/blob/87e1f7b5fdf5a138794121ca2965badc4a64b30a/src/wifi_manager.c#L1265

and this line

https://github.com/tonyp7/esp32-wifi-manager/blob/master/src/wifi_manager.c#L938

... done.

A configuration item WIFI_MANAGER_AP_ALWAYS_ON might be added in the future.

closed 2020-08-07 by tonyp7