tonyp7/esp32-wifi-manager #102: Solving ERR_CONNECTION_RESET error using ESP-IDF 3.3

EtienneMdv, 2020-08-06:
Hi @tonyp7,

Congrats for the amazing job! I have been using an old version of this library for my project which relies on ESP-IDF 3.3 (I use this version for compatibility reasons with the amazon-freertos library). This older version does not use the esp http server API but your own implementation. However, as it has been brought up before I sometimes get an ERR_CONNECTION_RESET error while trying to access the portal. Do you know how I could solve this problem (maybe only change the http_server.x files) without switching to ESP-IDF 4.x?

Best,
Etienne

tonyp7, 2020-08-06:
Hello Etienne,

Unfortunately no, I could never figure out what was wrong with my implementation. Moreover it crashes on 4.x which is what pushed me to go with the esp32-http-server component.

I do believe that getting requests from the AP or the STA and changing wifi mode messes with the way sockets handle data. In the newer version everytime you switch from AP to APSTA or vice versa the HTTP server is reset for this very reason. If you're keen I'd recommend you do that but that might involve quite a bit of code refactoring.

Thank you for the kind words!

EtienneMdv, 2020-08-06:
Thanks @tonyp7 for your quick response! I have a better understanding of the issue now. I shall give a try to reset the HTTP server when switching wifi modes.

Best,
Etienne

closed 2020-08-06 by EtienneMdv