tonyp7/esp32-wifi-manager #103: Unknown type name 'esp_event_handler_instance_t' / implicit declaration of function 'esp_event_handler_instance_register'

shaffenmeister, 2020-08-06:
### Prerequisites

- [x] I have written a descriptive issue title
- [x] I have verified that I am using the latest version of esp32-wifi-manager
- [x] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description
When compiling the default application or an own application using esp32-wifi-manager, the following error messages are issued:

`wifi_manager.c:892:5: error: unknown type name 'esp_event_handler_instance_t'; did you mean 'esp_event_handler_t'?`
`wifi_manager.c:894:21: error: implicit declaration of function 'esp_event_handler_instance_register'`

### Steps to Reproduce
1. Clone esp-idf 4.1 (branch release/v4.1 or branch v4.1-rc)
2. Clone esp32-wifi-manager as per documentation. Branch: v3.2
3. cd esp32-wifi-manager/examples/default_demo
4. source export.sh
5. idf.py build

### System Configuration
Ubuntu 20.04 AMD64
ESP IDF as specified above
esp32-wifi-manager v3.2

Additional information: Compilation works with the esp idf master .

tonyp7, 2020-08-07:
Very very interesting. Something didn’t make the cut in the 4.1 release candidate but it’s unclear what it is. 

Thank you for the report.

I’ll open an incident at esp-idf and see what they say.

tonyp7, 2020-08-07:
Update.

As per usual, the answer often lies in the documentation.

Looking at:
https://docs.espressif.com/projects/esp-idf/en/v4.1-rc/api-reference/system/esp_event.html

or
https://docs.espressif.com/projects/esp-idf/en/release-v4.1/api-reference/system/esp_event.html

It appears 4.1 does not support esp_event_handler_instance_register yet and this is a feature that will come in 4.2.

I will update the README accordingly. As 4.2 is yet to release esp32-wifi-manager will only work with the master tree for now.

tonyp7, 2020-08-07:
Updated documentation -- see https://github.com/tonyp7/esp32-wifi-manager/commit/87e1f7b5fdf5a138794121ca2965badc4a64b30a

Thank you again for the report

closed 2020-08-07 by tonyp7

kaizoku-oh, 2020-10-16:
I'm facing the same issue when working with ESP-IDF 4.1.
` error: unknown type name 'esp_event_handler_instance_t'; did you mean 'esp_event_handler_t'?`
Is there anyway to get around this build error while still using the 4.1 version.