tonyp7/esp32-wifi-manager #105: ESP32 doesn't save a password to configured access point and configuration web page hangs on connecting

pawelc06, 2020-08-19:
### Description
ESP32 doesn't save a password to configured access point and configuration web page hangs on connecting.
ESP 32 STA gets IP from DHCP.
After ESP32 reset it is visible in logs that configuration of access point was not saved.
I tried with at least 2 different access points with the sam effect.

### Steps to Reproduce
1. Erase the ESP32 device
2. Flash ESP32 with master branch(19.08.2020)
3. Reset device and enable setial monitor to observe the logs
4. Connect to esp32 wi fi access point
5. Open configuration web page
6. Select home router access point (in this case 'vault05') 
7. Enter password and click Join
8. Connection to vault05 screen displays permanently with pulsing circle

### System Configuration
Module used: ESP32 DEVKITV1
Visual Studio with ESP32 IDF 4.2
Windows 8.1

Logs from serial monitor after connecting:
I (43810) http_server: POST /connect.json
I (43810) http_server: ssid: vault05, password: XXXXXXXXXXXXXXX22
I (43810) wifi_manager: MESSAGE: ORDER_CONNECT_STA
I (44020) dns_server: Replying to DNS request for android.prod.cloud.netflix.com from 10.10.0.2
I (45310) http_server: GET /status.json
I (45470) http_server: GET /status.json
I (46620) wifi_manager: WIFI_EVENT_STA_CONNECTED
I (47620) esp_netif_handlers: sta ip: 192.168.50.219, mask: 255.255.255.0, gw: 192.168.50.1
I (47620) wifi_manager: IP_EVENT_STA_GOT_IP
I (47620) wifi_manager: WM_EVENT_STA_GOT_IP
I (47630) wifi_manager: Set STA IP String to: 192.168.50.219
I (47630) wifi_manager: About to save config to flash
I (50620) main: free heap: 204548
I (60620) main: free heap: 204548
I (70620) main: free heap: 204548
I (80620) main: free heap: 204548
I (90620) main: free heap: 204548
I (100620) main: free heap: 204548
I (110620) main: free heap: 204548
I (120620) main: free heap: 204548
I (130620) main: free heap: 204548
I (140620) main: free heap: 204548
I (150620) main: free heap: 204548
I (160620) main: free heap: 204760
I (170620) main: free heap: 204760

tonyp7, 2020-08-19:
Hello,

If you see the message 
"I (47630) wifi_manager: About to save config to flash"

It means you have entered wifi_manager_save_sta_config. What happens once you reset your ESP?

pawelc06, 2020-08-19:
Hi tonyp7,
As I wrote It reports "No saved wifi found on startup":
I (1520) http_server: Registering URI handlers
I (1520) wifi_manager: MESSAGE: ORDER_LOAD_AND_RESTORE_STA
I (1530) wifi_manager: No saved wifi found on startup. Starting access point.
I (1530) wifi_manager: MESSAGE: ORDER_START_AP
I (1550) wifi_manager: WIFI_EVENT_AP_START
I (1650) http_server: Registering URI handlers
I (1650) dns_server: DNS Server listening on 53/udp
I (10620) main: free heap: 209616

Certainly for some reason saving of configuration and further re-connection doesn't occur.
Also callback function is not invoked.
Do you know how to force verbose mode to have more logs from WiFi manager/IDF libraries?

Cheers,
Pawel

pawelc06, 2020-08-19:
Additional information:
After adding some debugs in wifi_manager.c, I identified where it blocks:
`if(wifi_manager_config_sta && nvs_sync_lock( portMAX_DELAY )){`

function **nvs_sync_lock** doesn't return true so it seems it can't get NVS semaphore.
I would appreciate any hints for me for further troubleshooting...

Pawel

tonyp7, 2020-08-20:
I could not reproduce your issue so I ran

```bash
idf.py erase_flash
```
to completely restart anew. Now I can see your issue. It doesn't come from the lock though -- still checking

tonyp7, 2020-08-20:
I found it... 

On the first run, if your flash memory is brand new, wifi_manager locks the mutex but doesn't release it because nvs_open does not resolve (the namespace has not been created yet).

I tripped on my own synchronization API. Good job me. Thanks for the reporting.

Could you give a try to the master branch?

closed 2020-08-20 by tonyp7

pawelc06, 2020-08-20:
Hi Tony,
Thank you for your investigation and fixing the issue.
I pulled master with your recent commit and I can confirm it works fine now.
Great work!
I've recently moved to ESP32 IDF as I want to learn FreeRTOS programming and your project was exactly what I needed as a starting point for WiFi provisioning.

Looking forward for future development.
Pawel

tonyp7, 2020-08-20:
Great to hear people moving with the idf and freeRTOS. It's an amazing environment.

you can checkout my other project https://github.com/tonyp7/esp32-nixie-clock which uses a lot more freeRTOS and esp-idf features

commit 16b07c7e (2020-08-20): fix #105
commit 235b63e2 (2020-08-20): Merge pull request #106 from tonyp7/bug-fixes

fix #105