tonyp7/esp32-wifi-manager #108: Crashing on MESSAGE: EVENT_STA_DISCONNECTED

Azmatron, 2020-09-16:
### Prerequisites

- [x] I have written a descriptive issue title
- [x] I have verified that I am using the latest version of esp32-wifi-manager
- [x] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description
Using esp32-wifi-manager, it works flawlessly until the ESP32 cannot connect to the wireless router. As soon as the wifi connection is not seen the ESP crashes. It seems to be some form of heap corruption and occurs when the disconnect message is added to the queue. I have tried on multiple devices and multiple applications, including stripped down application and even the examples app provided. All have the same behavior.

Is this reproducible or am I doing something stupid?

Below are logs from running the "default_demo" example

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7008
ho 0 tail 12 room 4
load:0x40078000,len:13860
load:0x40080400,len:3952
0x40080400: _init at ??:?

entry 0x40080678
I (31) boot: ESP-IDF v4.3-dev-907-g6c17e3a64-dirty 2nd stage bootloader
I (31) boot: compile time 20:49:49
I (32) boot: chip revision: 1
I (36) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1ecd0 (126160) map
I (164) esp_image: segment 1: paddr=0x0002ecf8 vaddr=0x3ffb0000 size=0x01320 (  4896) load
I (166) esp_image: segment 2: paddr=0x00030020 vaddr=0x400d0020 size=0x7b970 (506224) map
0x400d0020: _stext at ??:?

I (363) esp_image: segment 3: paddr=0x000ab998 vaddr=0x3ffb1320 size=0x027d8 ( 10200) load
I (367) esp_image: segment 4: paddr=0x000ae178 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at X:/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (370) esp_image: segment 5: paddr=0x000ae584 vaddr=0x40080404 size=0x15828 ( 88104) load
I (428) boot: Loaded app from partition at offset 0x10000
I (429) boot: Disabling RNG early entropy source...
I (440) cpu_start: cpu freq: 160
I (440) cpu_start: Pro cpu up.
I (440) cpu_start: Starting app cpu, entry point is 0x400813d4
0x400813d4: call_start_cpu1 at X:/esp-idf/components/esp_system/port/cpu_start.c:112

I (428) cpu_start: App cpu up.
I (458) cpu_start: Pro cpu start user code
I (459) cpu_start: Application information:
I (459) cpu_start: Project name:     wifi_manager
I (464) cpu_start: App version:      5e8b4c6-dirty
I (469) cpu_start: Compile time:     Sep 16 2020 20:47:24
I (476) cpu_start: ELF file SHA256:  2fa8800e99479cd5...
I (482) cpu_start: ESP-IDF:          v4.3-dev-907-g6c17e3a64-dirty
I (489) heap_init: Initializing. RAM available for dynamic allocation:
I (496) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (502) heap_init: At 3FFB80C0 len 00027F40 (159 KiB): DRAM
I (508) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (514) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (521) heap_init: At 40095C2C len 0000A3D4 (40 KiB): IRAM
I (528) spi_flash: detected chip: generic
I (532) spi_flash: flash io: dio
I (537) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (629) wifi_manager: Set STA IP String to: 0.0.0.0
I (639) main: free heap: 251892
I (649) system_api: Base MAC address is not set
I (649) system_api: read default base MAC address from EFUSE
I (669) wifi_init: rx ba win: 6
I (669) wifi_init: tcpip mbox: 32
I (669) wifi_init: udp mbox: 6
I (669) wifi_init: tcp mbox: 6
I (669) wifi_init: tcp tx win: 5744
I (679) wifi_init: tcp rx win: 5744
I (679) wifi_init: tcp mss: 1440
I (689) wifi_init: WiFi IRAM OP enabled
I (689) wifi_init: WiFi RX IRAM OP enabled
I (1559) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 0
I (1569) wifi_manager:
I (1569) http_server: Registering URI handlers
I (1579) wifi_manager: wifi_manager_fetch_wifi_sta_config: ssid:AzCOM_MICRO password:xxxxxxxx
I (1579) wifi_manager: Saved wifi found on startup. Will attempt to connect.
I (1589) wifi_manager: MESSAGE: ORDER_CONNECT_STA
I (3649) wifi_manager: WIFI_EVENT_STA_DISCONNECTED
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008aecb  PS      : 0x00060233  A0      : 0x8008b42c  A1      : 0x3ffbf420
0x4008aecb: is_free at X:/esp-idf/components/heap/multi_heap.c:138
 (inlined by) get_prev_free_block at X:/esp-idf/components/heap/multi_heap.c:194

A2      : 0x004f5243  A3      : 0x3ffc926c  A4      : 0x3ffbc178  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x00000001
A10     : 0xfffffffc  A11     : 0x00000000  A12     : 0x3ffc9238  A13     : 0x00060223
A14     : 0x00000001  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x004f5243  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x4008aec8:0x3ffbf420 0x4008b429:0x3ffbf440 0x40081986:0x3ffbf460 0x4008e8a5:0x3ffbf480 0x4014ad5e:0x3ffbf4a0 0x4014add5:0x3ffbf4e0 0x40087a51:0x3ffbf500
0x4008aec8: get_prev_free_block at X:/esp-idf/components/heap/multi_heap.c:191 (discriminator 1)

0x4008b429: multi_heap_free_impl at X:/esp-idf/components/heap/multi_heap.c:547

0x40081986: heap_caps_free at X:/esp-idf/components/heap/heap_caps.c:305

0x4008e8a5: free at X:/esp-idf/components/newlib/heap.c:47

0x4014ad5e: post_instance_delete at X:/esp-idf/components/esp_event/esp_event.c:436
 (inlined by) esp_event_loop_run at X:/esp-idf/components/esp_event/esp_event.c:604

0x4014add5: esp_event_loop_run_task at X:/esp-idf/components/esp_event/esp_event.c:115

0x40087a51: vPortTaskWrapper at X:/esp-idf/components/freertos/xtensa/port.c:169

```

### Steps to Reproduce
1. Load default_demo
2. Connect to AP using and configure local wifi
3. Remove power from Wifi router to simulate a  WIFI_EVENT_STA_DISCONNECTED
4. ESP crashes with "Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled."

### System Configuration
Building on Windows 10
Latest esp-idf (ESP-IDF v4.3-dev-907)
Lastest esp32-wifi-manager

Azmatron, 2020-09-16:
As a further note. I tried reverting back to ESP-IDF 4.2 with the same result

tonyp7, 2020-09-17:
Thank you for the detailed report. Hopefully it should be an easy one to reproduce on my end. I’ll check this weekend.

Azmatron, 2020-09-17:
Hi thank you for the quick response (and sharing this really cool app). I think that I have found the issue.

In the wifi_manager_event_handler function at the WIFI_EVENT_STA_DISCONNECTED case, the code is allocating memory for the disconnected param but it is using the variable rather than type to determine size.

`wifi_event_sta_disconnected_t*` wifi_event_sta_disconnected = (wifi_event_sta_disconnected_t*)malloc(sizeof(wifi_event_sta_disconnected));`

I believe rather it should be 

`wifi_event_sta_disconnected_t*` wifi_event_sta_disconnected = (wifi_event_sta_disconnected_t*)malloc(sizeof(wifi_event_sta_disconnected**_t**));`

I have added this change locally and it seems to now be behaving correctly with no heap corruption.

Please let me know your thoughts

tonyp7, 2020-09-17:
Yeah that's a big oops. I have checked the code and the rest of the params are ok.

eg

```c
wifi_event_sta_scan_done_t* event_sta_scan_done = (wifi_event_sta_scan_done_t*)malloc(sizeof(wifi_event_sta_scan_done_t));
```

and

```c
ip_event_got_ip_t* ip_event_got_ip = (ip_event_got_ip_t*)malloc(sizeof(ip_event_got_ip_t));
```

commit 1bc7189b (2020-10-04): Fix #108
commit 75c92533 (2020-10-04): Merge pull request #111 from tonyp7/bug-fixes

Fix #108
closed 2020-10-04 by tonyp7