tonyp7/esp32-wifi-manager #110: crash on connect

perigoso, 2020-10-01:
### Prerequisites

- [x] I have written a descriptive issue title
- [x] I have verified that I am using the latest version of esp32-wifi-manager
- [x] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description
I'm not sure if this is a bug, or just something i did wrong.

When the esp boots it goes to the wi-fi manager just fine, but when i try to add a connection it seems to crash, the culprit for the error seems to be `esp_wifi_scan_start(&scan_config, false)` at line 1020 (in master branch, 981 on the commit the project is using, but i have tried both versions without success), but I am not sure what to make from it.

### Steps to Reproduce

- clone [esp32-nixie-clock](https://github.com/tonyp7/esp32-nixie-clock)
- compile with latest master branch of esp-idf (ESP-IDF v4.3-dev-1197-g8bc19ba89)
- erase flash and flash esp
- connect to AP
- add new connection

### System Configuration

I am trying to use the code from [esp32-nixie-clock](https://github.com/tonyp7/esp32-nixie-clock) (I know it's a different project but the problem seems to be the wi-fi manager), with custom hardware I made [here](https://github.com/tonyp7/esp32-nixie-clock), I don't see a reason to suspect hardware.

Log:
```
I (12) boot: ESP-IDF v4.3-dev-1197-g8bc19ba89 2nd stage bootloader
I (12) boot: compile time 02:08:04
I (12) boot: chip revision: 1
I (16) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (23) boot.esp32: SPI Speed      : 40MHz
I (27) boot.esp32: SPI Mode       : DIO
I (32) boot.esp32: SPI Flash Size : 2MB
I (36) boot: Enabling RNG early entropy source...
I (42) boot: Partition Table:
I (45) boot: ## Label            Usage          Type ST Offset   Length
I (53) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (60) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (68) boot:  2 factory          factory app      00 00 00010000 00100000
I (75) boot: End of partition table
I (79) boot_comm: chip revision: 1, min. application chip revision: 0
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x33d1c (212252) map
I (176) esp_image: segment 1: paddr=0x00043d44 vaddr=0x3ffb0000 size=0x03b18 ( 15128) load
I (183) esp_image: segment 2: paddr=0x00047864 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /opt/esp-idf-sdk/components/freertos/xtensa/xtensa_vectors.S:1730

I (183) esp_image: segment 3: paddr=0x00047c70 vaddr=0x40080404 size=0x083a8 ( 33704) load
I (207) esp_image: segment 4: paddr=0x00050020 vaddr=0x400d0020 size=0xa12b8 (660152) map
0x400d0020: _stext at ??:?

I (458) esp_image: segment 5: paddr=0x000f12e0 vaddr=0x400887ac size=0x10134 ( 65844) load
0x400887ac: prvCopyItemAllowSplit at /opt/esp-idf-sdk/components/esp_ringbuf/ringbuf.c:436

I (501) boot: Loaded app from partition at offset 0x10000
I (501) boot: Disabling RNG early entropy source...
I (512) cpu_start: cpu freq: 160
I (512) cpu_start: Pro cpu up.
I (512) cpu_start: Starting app cpu, entry point is 0x40081588
0x40081588: call_start_cpu1 at /opt/esp-idf-sdk/components/esp_system/port/cpu_start.c:112

I (0) cpu_start: App cpu up.
I (533) cpu_start: Pro cpu start user code
I (533) cpu_start: Application information:
I (533) cpu_start: Project name:     nixieclock
I (538) cpu_start: App version:      7e67029
I (543) cpu_start: Compile time:     Oct  1 2020 02:07:56
I (549) cpu_start: ELF file SHA256:  46b55632506b1e3a...
I (555) cpu_start: ESP-IDF:          v4.3-dev-1197-g8bc19ba89
I (562) heap_init: Initializing. RAM available for dynamic allocation:
I (569) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (575) heap_init: At 3FFB83E8 len 00027C18 (159 KiB): DRAM
[... 86 lines cut ...]

file: "../components/esp32-wifi-manager/src/wifi_manager.c" line 981
func: wifi_manager
expression: esp_wifi_scan_start(&scan_config, false)

abort() was called at PC 0x40087fc3 on core 0
0x40087fc3: _esp_error_check_failed at /opt/esp-idf-sdk/components/esp_common/src/esp_err.c:42

Backtrace:0x4008910f:0x3ffbf2d0 0x40089869:0x3ffbf2f0 0x40091fd2:0x3ffbf310 0x40087fc3:0x3ffbf380 0x400dd257:0x3ffbf3a0 0x40089871:0x3ffbf530
0x4008910f: panic_abort at /opt/esp-idf-sdk/components/esp_system/panic.c:349

0x40089869: esp_system_abort at /opt/esp-idf-sdk/components/esp_system/system_api.c:104

0x40091fd2: abort at /opt/esp-idf-sdk/components/newlib/abort.c:46

0x40087fc3: _esp_error_check_failed at /opt/esp-idf-sdk/components/esp_common/src/esp_err.c:42

0x400dd257: wifi_manager at /home/gimbas/Development/esp32-nixie-clock/nixie_clock_code/build/../components/esp32-wifi-manager/src/wifi_manager.c:981 (discriminator 1)

0x40089871: vPortTaskWrapper at /opt/esp-idf-sdk/components/freertos/xtensa/port.c:169

ELF file SHA256: 46b55632506b1e3a

Rebooting...
```

MPC-BlackBox, 2020-10-02:
Hi,

I had the same issue. Not sure what is causing it but I managed to get round the issue by changing line 1020 to
//					ESP_ERROR_CHECK(esp_wifi_scan_start(&scan_config, false));
					esp_err_t error = esp_wifi_scan_start(&scan_config, false);
					if (error!=ESP_OK)
						ESP_LOGI(TAG, "MESSAGE: WIFI SCAN FAILED WITH CODE %d",error);

esp_wifi_scan_start returns -1 or ESP_FAIL so the error check fails. The API does not give any reasons why esp_wifi_scan_stat should return -1.

The code ignores errors which works as a quick fix until something more meaningful comes along.

MPC

perigoso, 2020-10-02:
@MPC-BlackBox I tried that but my esp just started crashing after with some heap error

tonyp7, 2020-10-04:
The problem is that ESP_FAIL isn't a value that is documented for esp_wifi_scan_start. I'm wondering if this is something going on with 4.3 since noone ever reported this kind of issues and now I have two in a row.

Also is the sourcecode for esp_wifi.h not public? I wanted to look at what esp_wifi_scan_start but I cannot find the definition anywhere!

tonyp7, 2020-10-04:
Also @perigoso just for context esp32-nixie-clock was last compiled on a 4.2 beta master from esp-idf and I can switch networks without issues.

I'll have to update my repos and try to get this to crash on a dev board. 

Everytime esp-idf is updated it seems like something breaks...

perigoso, 2020-10-04:
I also tried to look for the function and couldn't find anything, so it's hard to know what is really going for it to return -1.

I intend to test with esp-idf 4.4beta1 and see if it works.

perigoso, 2020-10-08:
@tonyp7 maybe I should try a specific commit of the v4.2 esp-idf, could this be related to my system (as in my pc) configuration at all?

```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (13) boot: ESP-IDF v4.2-beta1-dirty 2nd stage bootloader
I (13) boot: compile time 23:19:00
I (13) boot: chip revision: 1
I (16) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (23) qio_mode: Enabling default flash chip QIO
I (28) boot.esp32: SPI Speed      : 40MHz
I (33) boot.esp32: SPI Mode       : QIO
I (37) boot.esp32: SPI Flash Size : 4MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (85) boot_comm: chip revision: 1, min. application chip revision: 0
I (92) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x33b70 (211824) map
I (170) esp_image: segment 1: paddr=0x00043b98 vaddr=0x3ffb0000 size=0x03b6c ( 15212) load
I (176) esp_image: segment 2: paddr=0x0004770c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /opt/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (177) esp_image: segment 3: paddr=0x00047b18 vaddr=0x40080404 size=0x08500 ( 34048) load
I (199) esp_image: segment 4: paddr=0x00050020 vaddr=0x400d0020 size=0xa12b4 (660148) map
0x400d0020: _stext at ??:?

I (415) esp_image: segment 5: paddr=0x000f12dc vaddr=0x40088904 size=0x1058c ( 66956) load
0x40088904: prvReceiveGeneric at /opt/esp-idf/components/esp_ringbuf/ringbuf.c:763

I (454) boot: Loaded app from partition at offset 0x10000
I (454) boot: Disabling RNG early entropy source...
I (455) cpu_start: Pro cpu up.
I (458) cpu_start: Application information:
I (463) cpu_start: Project name:     nixieclock
I (468) cpu_start: App version:      b100565-dirty
I (474) cpu_start: Compile time:     Oct  8 2020 23:18:54
I (480) cpu_start: ELF file SHA256:  3ad9c1c9860cf74a...
I (486) cpu_start: ESP-IDF:          v4.2-beta1-dirty
I (492) cpu_start: Starting app cpu, entry point is 0x40081a6c
0x40081a6c: call_start_cpu1 at /opt/esp-idf/components/esp32/cpu_start.c:282

I (0) cpu_start: App cpu up.
I (502) heap_init: Initializing. RAM available for dynamic allocation:
I (509) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (515) heap_init: At 3FFB9748 len 000268B8 (154 KiB): DRAM
I (521) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (528) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (534) heap_init: At 40098E90 len 00007170 (28 KiB): IRAM
I (540) cpu_start: Pro cpu start user code
W (706) clk: 32 kHz XTAL not found, switching to internal 150 kHz oscillator
I (716) spi_flash: detected chip: generic
I (716) spi_flash: flash io: qio
I (716) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (827) wifi_manager: Set STA IP String to: 0.0.0.0
I (837) DS3231: READ: YEAR:0 MONTH:0 DAY:1 [WDAY:0] - 0:4:39
I (837) clock: The current RTC time is: Sun Jan  1 00:04:39 1900
I (837) clock: clock_save_config_task started
I (837) clock: Getting config from NVS memory
I (847) system_api: Base MAC address is not set
I (847) gpio: GPIO[4]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:1 
I (857) system_api: read default base MAC address from EFUSE
I (857) neopix: Received R:0 G:0 B:0
I (867) clock: CLOCK_MESSAGE_SLEEPMODE_CONFIG
I (837) main: free heap: 198400
I (1757) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 0
[... 43 lines cut ...]

0x40081551: heap_caps_malloc_default at /opt/esp-idf/components/heap/heap_caps.c:177

0x400907a5: malloc at /opt/esp-idf/components/newlib/heap.c:32

0x400e97cd: mem_malloc at /opt/esp-idf/components/lwip/lwip/src/core/mem.c:237

0x400e9843: do_memp_malloc_pool at /opt/esp-idf/components/lwip/lwip/src/core/memp.c:254

0x400e98b4: memp_malloc at /opt/esp-idf/components/lwip/lwip/src/core/memp.c:350 (discriminator 2)

0x400f0995: sys_timeout_abs at /opt/esp-idf/components/lwip/lwip/src/core/timeouts.c:198

0x400f0a4e: sys_timeout at /opt/esp-idf/components/lwip/lwip/src/core/timeouts.c:314

0x400f0aa2: tcpip_tcp_timer at /opt/esp-idf/components/lwip/lwip/src/core/timeouts.c:163

0x400f0b49: sys_check_timeouts at /opt/esp-idf/components/lwip/lwip/src/core/timeouts.c:400

0x400e7f96: tcpip_timeouts_mbox_fetch at /opt/esp-idf/components/lwip/lwip/src/api/tcpip.c:115

0x400e8048: tcpip_thread at /opt/esp-idf/components/lwip/lwip/src/api/tcpip.c:148

0x40089775: vPortTaskWrapper at /opt/esp-idf/components/freertos/xtensa/port.c:143
```

tonyp7, 2020-10-09:
@perigoso that's a different issue that was fixed with the latest commit in the master branch ( #108 )

perigoso, 2020-10-14:
@tonyp7 ok, the heap crash was indeed fixed on the master branch, but I am still unable to connect even with esp-idf `4.2beta1`, `esp_wifi_scan_start(&scan_config, false)` still fails with `-1`

perigoso, 2020-10-26:
@tonyp7 hey, have you had any time to try and reproduce this issue? I got around it for now by having the credentials hard-coded once, further reprograms work by fetching from memory as long as i don't erase flash.
If I can do anything to help I'd be happy to.

mjuhanne, 2020-10-26:
**Found the reason for crash**: Scan cannot be started when simultaneously trying to connect to station (esp_wifi_scan_start returns ESP_ERR_WIFI_STATE, which ESP_ERROR_CHECK doesn't handle well).  I first tried to check additional flags (WIFI_MANAGER_REQUEST_STA_CONNECT_BIT and WIFI_MANAGER_REQUEST_RESTORE_STA_BIT) but I still got occasional race conditions, so I check for ESP_ERR_WIFI_STATE separately anyway and let ESP_ERROR_CHECK() handle the rest.

```
			case WM_ORDER_START_WIFI_SCAN:
				ESP_LOGD(TAG, "MESSAGE: ORDER_START_WIFI_SCAN");

				/* if a scan is already in progress this message is simply ignored thanks to the WIFI_MANAGER_SCAN_BIT uxBit */
				uxBits = xEventGroupGetBits(wifi_manager_event_group);
				if (! (uxBits & WIFI_MANAGER_SCAN_BIT) ){
					if (! (uxBits & ( WIFI_MANAGER_REQUEST_STA_CONNECT_BIT | WIFI_MANAGER_REQUEST_RESTORE_STA_BIT) ) ) {
						esp_err_t res = esp_wifi_scan_start(&scan_config, false);
						if (res==ESP_ERR_WIFI_STATE) {
							// This might happen when connect retry is attempted while AP scan starts.. Just ignore it
							ESP_LOGE(TAG,"Wifi still in connect mode whan starting scan!");
						} else {
							// handle other errors
							ESP_ERROR_CHECK(res);
							xEventGroupSetBits(wifi_manager_event_group, WIFI_MANAGER_SCAN_BIT);
						}
					} else {
						// Cannot scan while connecting to AP
					}
				}
```

tonyp7, 2020-10-27:
... Wait you mean the -1 reported by ESP_ERROR_CHECK is wrong? That's a great find!

gagank1, 2021-02-01:
@mjuhanne @tonyp7 This fix doesn't address the case where `ESP_FAIL` (-1) is returned though. Is it safe to ignore it along with ESP_ERR_WIFI_STATE?

sullij, 2021-08-08:
> **Found the reason for crash**: Scan cannot be started when simultaneously trying to connect to station (esp_wifi_scan_start returns ESP_ERR_WIFI_STATE, which ESP_ERROR_CHECK doesn't handle well). I [...]

I think I had solved a similar issue a many months ago and I just briefly read through this thread and it seems to be a similar issue. I solved by adjusting based on the below:

				case ORDER_START_WIFI_SCAN:
					ESP_LOGD(TAG, "MESSAGE: ORDER_START_WIFI_SCAN");

					uxBits = xEventGroupGetBits(wifi_manager_event_group);
					// If WIFI_MANAGER_REQUEST_ORDER_CONNECT_STA_BIT is SET, do NOT start scanning function as this will cause problems connecting to the external AP
					if(!(uxBits & WIFI_MANAGER_REQUEST_ORDER_CONNECT_STA_BIT))
					{
						// if a scan is already in progress this message is ignored thanks to the WIFI_MANAGER_SCAN_BIT uxBit
						if(!(uxBits & WIFI_MANAGER_SCAN_BIT))
						{
							xEventGroupSetBits(wifi_manager_event_group, WIFI_MANAGER_SCAN_BIT);
							// Including esp_wifi_disconnect as per Espressif recommendation (solves another scanning issue)
							ESP_ERROR_CHECK(esp_wifi_disconnect());
							ESP_ERROR_CHECK(esp_wifi_scan_start(&scan_config, true));
						}
					}
					break;

There are two important bits there. 

First: I added the "WIFI_MANAGER_REQUEST_ORDER_CONNECT_STA_BIT", which I set under the "ORDER_CONNECT_STA" case and clear under the "EVENT_STA_DISCONNECTED" case. 

Second: I added ESP_ERROR_CHECK(esp_wifi_disconnect()); as per an Espressif recommendation I found somewhere (I can't remember where, it's been a very long time since solving this).

I hope that relates to your issue and helps out - if not, my mistake!

xobs, 2022-04-22:
Thanks, @MPC-BlackBox. I've incorporated this into a patch at https://github.com/xobs/esp32-wifi-manager/commit/a0bc1bca0dbc0780dbb993806b1db8f35fd430a7 for those of you who are maintaining your own forks now.