tonyp7/esp32-wifi-manager #112: Can't build example using esp-idf v4.1

kaizoku-oh, 2020-10-16:
### Prerequisites

- [x] I have written a descriptive issue title
- [x] I have verified that I am using the latest version of esp32-wifi-manager
- [x] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description
I'm facing the same issue as in #103 when working with ESP-IDF 4.1.
` error: unknown type name 'esp_event_handler_instance_t'; did you mean 'esp_event_handler_t'?`
**Is there anyway to get around this build error while still using the 4.1 version?**

### Steps to Reproduce

1. install platformio extention on vscode
2. create a new project for esp32 using the esp-idf framework
3. create a components folder under the root of the project and copy the wifi-manager into it (tried master and other tags)
4. build

### System Configuration

- Debian 10 x64
- vscode
- platformio

tonyp7, 2020-10-16:
Hello,

This is normal, this code only compile with 4.2 and up as per the requirements:

https://github.com/tonyp7/esp32-wifi-manager#requirements

kaizoku-oh, 2020-10-16:
Are you considering to support a stable release in the future? because v4.2 is only pre-release beta now contrary to v4.1 which will be supported until February 2023.

tonyp7, 2020-10-17:
Definitely not. 4.2 should hit RC stage fairly soon and the master tree is already in 4.3. There’s no point holding back the codebase at this stage.

closed 2020-10-17 by kaizoku-oh