tonyp7/esp32-wifi-manager #116: Source `style.css.S` not found error while building

honestech74, 2020-10-22:
### Description
I am using ESP-IDF v4 and PlatformIO and got the following error while building the project. I have added this repository to my components directory and set EXTRA_COMPONENTS, and then copy the whole code of default_demo/main/user_main.c to my project source.

```
> Executing task: C:\Users\hones\.platformio\penv\Scripts\platformio.exe run <

Processing esp32cam (platform: espressif32; board: esp32cam; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32cam.html
PLATFORM: Espressif 32 (2.0.0) > AI Thinker ESP32-CAM
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)    
PACKAGES:
 - framework-espidf 3.40100.200827 (4.1.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Warning: the 'src_filter' option cannot be used with ESP-IDF. Select source files to build in the project CMakeLists.txt file.

Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32cam\esp-idf\src\main.c.o
Generating LD script .pio\build\esp32cam\esp32_out.ld
Generating partitions .pio\build\esp32cam\partitions.bin
Compiling .pio\build\esp32cam\esp-idf\app_trace\app_trace.c.o
Compiling .pio\build\esp32cam\esp-idf\app_trace\app_trace_util.c.o
Compiling .pio\build\esp32cam\esp-idf\app_trace\host_file_io.c.o
Compiling .pio\build\esp32cam\esp-idf\app_trace\gcov\gcov_rtio.c.o
Compiling .pio\build\esp32cam\esp-idf\app_update\esp_ota_ops.c.o
Compiling .pio\build\esp32cam\esp-idf\app_update\esp_app_desc.c.o
Compiling .pio\build\esp32cam\esp-idf\asio\asio\asio\src\asio.cpp.o
Archiving .pio\build\esp32cam\esp-idf\app_update\libapp_update.a
Archiving .pio\build\esp32cam\esp-idf\app_trace\libapp_trace.a
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\bootloader_clock.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\bootloader_common.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\bootloader_flash.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\bootloader_random.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\bootloader_utility.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\esp_image_format.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\flash_partitions.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\flash_qio_mode.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\bootloader_flash_config_esp32.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\bootloader_efuse_esp32.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\flash_encrypt.c.o
Compiling .pio\build\esp32cam\esp-idf\bootloader_support\src\idf\bootloader_sha.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cborencoder_close_container_checked.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cborencoder.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cborerrorstrings.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cborparser_dup_string.c.o
Archiving .pio\build\esp32cam\esp-idf\bootloader_support\libbootloader_support.a
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cborparser.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cborpretty_stdio.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cborpretty.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cbortojson.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\cborvalidation.c.o
Compiling .pio\build\esp32cam\esp-idf\cbor\tinycbor\src\open_memstream.c.o
Compiling .pio\build\esp32cam\esp-idf\coap\libcoap\src\address.c.o
Compiling .pio\build\esp32cam\esp-idf\coap\libcoap\src\async.c.o
Compiling .pio\build\esp32cam\esp-idf\coap\libcoap\src\block.c.o
Archiving .pio\build\esp32cam\esp-idf\cbor\libcbor.a
[... 87 lines cut ...]
Compiling .pio\build\esp32cam\esp-idf\esp32\task_wdt.c.o
Compiling .pio\build\esp32cam\esp-idf\esp32-wifi-manager\src\dns_server.c.o
Compiling .pio\build\esp32cam\esp-idf\esp32-wifi-manager\src\http_app.c.o
Compiling .pio\build\esp32cam\esp-idf\esp32-wifi-manager\src\json.c.o
Compiling .pio\build\esp32cam\esp-idf\esp32-wifi-manager\src\nvs_sync.c.o
Compiling .pio\build\esp32cam\esp-idf\esp32-wifi-manager\src\wifi_manager.c.o
Archiving .pio\build\esp32cam\esp-idf\esp32\libesp32.a
*** [.pio\build\esp32cam\.pio\build\esp32cam\style.css.S.o] Source '.pio\build\esp32cam\style.css.S' not found, needed by target '.pio\build\esp32cam\.pio\build\esp32cam\style.css.S.o'.
```

### Steps to Reproduce
- Make your project using ESP-IDF framework in PlatformIO.
- Copy this repository to components directory
- Add the EXTRA_COMPONENTS to your project
    `set(EXTRA_COMPONENTS components/)`
- Build your project using **Build** button of PlatformIO or CLI
    `pio run`

### System Configuration

- Environment:

OS: Windows 10 x64
IDE: PlatformIO
Tools: ESP-IDF v4.1

honestech74, 2020-10-22:
I have added the following configuration to `platformio.ini`, then an error has fixed.
```
board_build.embed_txtfiles = 
  components/esp32-wifi-manager/src/index.html
  components/esp32-wifi-manager/src/style.css
  components/esp32-wifi-manager/src/code.js
```

closed 2020-10-22 by honestech74

kaizoku-oh, 2020-11-01:
Hi, can share how you compiled it on platformio knowing that this project only compiles with esp-idf 4.2 and above, and platformio is using esp-idf v4.1

gagank1, 2021-01-31:
#### Fix for platformio/platform-espressif32 v3.0.0 and ESP-IDF v4.2

I had the same problem but with this fix the portal is static and the js doesn't work. This is because Platformio adds an extra null char at the end of the files, breaking the scripts. This is the fix for wifi-manager release v3.3.1 on Platformio v3.0.0.

Add this to your platform.ini:
```
board_build.embed_txtfiles = 
  components/esp32-wifi-manager/src/index.html
  components/esp32-wifi-manager/src/style.css
  components/esp32-wifi-manager/src/code.js
```

Then modify these 3 lines in http_app.c:

252: `httpd_resp_send(req, (char*)index_html_start, index_html_end - index_html_start);`
258: `httpd_resp_send(req, (char*)code_js_start, code_js_end - code_js_start);`
265: `httpd_resp_send(req, (char*)style_css_start, style_css_end - style_css_start);`

to
```
httpd_resp_send(req, (char*)index_html_start, strlen((char*)index_html_start));
httpd_resp_send(req, (char*)code_js_start, strlen((char*)code_js_start));
httpd_resp_send(req, (char*)style_css_start, strlen((char*)style_css_start));
```

R4k4210, 2022-03-30:
> #### Fix for platformio/platform-espressif32 v3.0.0 and ESP-IDF v4.2
> I had the same problem but with this fix alone the portal is static and the js doesn't work. This is because Platformio adds an extra null char at the end of the files, breaking the scripts. For anyone else reading, this is the full fix for wifi-manager release v3.3.1 on Platformio v3.0.0.
> 
> Add this to your platform.ini:
> 
> ```
> board_build.embed_txtfiles = 
>   components/esp32-wifi-manager/src/index.html
>   components/esp32-wifi-manager/src/style.css
>   components/esp32-wifi-manager/src/code.js
> ```
> 
> Then modify these 3 lines in http_app.c:
> 
> 252: `httpd_resp_send(req, (char*)index_html_start, index_html_end - index_html_start);` 258: `httpd_resp_send(req, (char*)code_js_start, code_js_end - code_js_start);` 265: `httpd_resp_send(req, (char*)style_css_start, style_css_end - style_css_start);`
> 
> to
> 
> ```
> httpd_resp_send(req, (char*)index_html_start, strlen((char*)index_html_start));
> httpd_resp_send(req, (char*)code_js_start, strlen((char*)code_js_start));
> httpd_resp_send(req, (char*)style_css_start, strlen((char*)style_css_start));
> ```

This fix should be in readme for Platformio users!! Thanks a lot!

thebitguru, 2022-06-26:
@gagank1 thank you for pointing in the right direction.  The NULL issue can be solved by using [`embed_files` instead of `embed_txtfiles`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#embedding-binary-data).  Doing this eliminates the need to change http_app.c.

```
board_build.embed_files = 
  components/esp32-wifi-manager/src/index.html
  components/esp32-wifi-manager/src/style.css
  components/esp32-wifi-manager/src/code.js
```