tonyp7/esp32-wifi-manager #126: Error while cmake a program

monhack, 2021-03-16:
Hi @everyone

I want to ask some question about cmake. What is payload for cmake this program?

I try to cmake but not success.

```
CMake Error at CMakeLists.txt:5 (include):
  include could not find load file:

    /tools/cmake/project.cmake

CMake Warning at /home/monhack/esp/esp-idf/tools/cmake/idf.cmake:24 (message):
  IDF_PATH environment variable not found.  Setting IDF_PATH to
  '/home/monhack/esp/esp-idf'.
Call Stack (most recent call first):
  /home/monhack/esp/esp-idf/tools/cmake/project.cmake:7 (include)
  wifi_1/CMakeFiles/3.18.1/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:7 (project)

-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /home/monhack/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/monhack/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - broken
CMake Error at /usr/local/share/cmake-3.18/Modules/CMakeTestCCompiler.cmake:66 (message):
  The C compiler

    "/home/monhack/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/monhack/Documents/Kerjaan/esp32-wifi-manager/examples/default_demo/wifi_1/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/home/monhack/.espressif/tools/ninja cmTC_90512/fast && ninja: error: loading 'build.ninja': No such file or directory
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:7 (project)

-- Configuring incomplete, errors occurred!
See also "/home/monhack/Documents/Kerjaan/esp32-wifi-manager/examples/default_demo/wifi_1/CMakeFiles/CMakeOutput.log".
See also "/home/monhack/Documents/Kerjaan/esp32-wifi-manager/examples/default_demo/wifi_1/CMakeFiles/CMakeError.log".
```

hanhdt, 2021-03-17:
Why don't you use [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/#) to build project?
```
$idf.py build
```

monhack, 2021-04-06:
Thanks @hanhdt,

i follow you instruction, but error

```
/home/pckharisma-2/esp/esp-idf/tools/idf.py build flash monitor -p /dev/ttyUSB0
Setting IDF_PATH environment variable: /home/pckharisma-2/esp/esp-idf
WARNING: Support for Python 2 is deprecated and will be removed in future versions.
The following Python requirements are not satisfied:
gdbgui==0.13.2.0
pygdbmi<=0.9.0.2
python-socketio<5; python_version<="2.7"
kconfiglib==13.7.1
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: /usr/bin/python
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: /home/pckharisma-2/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/pckharisma-2/esp/xtensa-esp32-elf/bin

ESP-IDF v4.4-dev-4-g73db14240-dirty
```

monhack, 2021-04-13:
finally solved

closed 2021-04-13 by monhack