tonyp7/esp32-wifi-manager #127: httpd_parse: parse_block: request URI/header too long

hackaprende, 2021-03-24:
### Prerequisites

- [x] I have written a descriptive issue title
- [x] I have verified that I am using the latest version of esp32-wifi-manager
- [ x] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description
When I run the project I get this error

[image]

The same is displayed in the browser

### Steps to Reproduce

- Add esp32-wifi-manager as component
- Call 	wifi_manager_start(); in app_main
- Connect to esp32
- "Header fields are too long for server to interpret" error appears

### System Configuration

- Environment (Operating system, version and so on): Esp32 devkitC with IDF V4.2, in MacOS
- Additional information:

hackaprende, 2021-03-24:
It happens in brave and duck duck go browsers in Android. It works well for Chrome

closed 2021-03-24 by hackaprende

daverathbone, 2022-01-30:
Just a lot of pain to find you use J and K keys for up down arrow - Make it harder!!!!!