tonyp7/esp32-wifi-manager #128: Web interface is not showing any AP

kaizoku-oh, 2021-03-25:
### Prerequisites

- [x] I have written a descriptive issue title
- [x] I have verified that I am using the latest version of esp32-wifi-manager
- [x] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description

I'm trying to use wifi manager but the Web interface is not showing any AP knowing that there are close nearby APs

### Steps to Reproduce

``` C

void wifi_manager_sta_got_ip_cb(void *pvParameter)
{
  ip_event_got_ip_t *pstIPAddr;
  char tcIpString[sizeof("xxx.xxx.xxx.xxx")];

  pstIPAddr = (ip_event_got_ip_t*)pvParameter;
  /* transform IP to human readable string */
  esp_ip4addr_ntoa(&pstIPAddr->ip_info.ip, tcIpString, IP4ADDR_STRLEN_MAX);
  ESP_LOGI(TAG, "I have a connection and my IP is %s!", tcIpString);
}

void wifi_manager_sta_discon_cb(void *pvParameter)
{
  wifi_event_sta_disconnected_t *pstWifiDiscon;

  pstWifiDiscon = (wifi_event_sta_disconnected_t*)pvParameter;
  ESP_LOGW(TAG, "Station disconnected with reason code: %d", pstWifiDiscon->reason);
}

/* start the wifi manager */
wifi_manager_start();
/* register a callback to be called when station is connected to AP */
wifi_manager_set_callback(WM_EVENT_STA_GOT_IP, &wifi_manager_sta_got_ip_cb);
/* register a callback to be called when station is disconnected from AP */
wifi_manager_set_callback(WM_EVENT_STA_DISCONNECTED, &wifi_manager_sta_discon_cb);

```

[image]

### System Configuration

ESP32 board

- Environment (Operating system, version and so on): Windows 10
- Additional information:

mtetreault, 2021-05-29:
Could you confirm that there are 2.4Ghz wifi network around you? The esp32 does not support 5Ghz wifi.

Otherwise, do you have a branch where I could take a look at the code, if you are using the default_demo as I think you are, it should work just fine as it is.

kaizoku-oh, 2021-05-29:
Yes there are 2.4GHz wifi network arround me.
Here is a the example I'm using: https://github.com/kaizoku-oh/pio-wifi-manager-example

mtetreault, 2021-05-29:
I've sent you a PR on your repo, everything builds and run on my side. I have no experiences with platformIO so there might be some differences. With my normal ESP-IDF setup, the project was not building it was giving me some errors.

I've built using the ESP-IDF docker and tested on ESP32-WROOM-32E.

Let me know how it goes.

JackHuang021, 2024-04-30:
set `board_build.embed_files` instead of `board_build.embed_txtfiles` in platformio.ini
```diff
diff --git a/platformio.ini b/platformio.ini
index 4f1498a..f8bdb0b 100644
--- a/platformio.ini
+++ b/platformio.ini
@@ -14,8 +14,9 @@ board = nodemcu-32s
 framework = espidf

 monitor_speed = 115200

-board_build.embed_txtfiles =
+board_build.embed_files =
   components/esp32-wifi-manager/src/index.html
   components/esp32-wifi-manager/src/code.js
   components/esp32-wifi-manager/src/style.css
```