tonyp7/esp32-wifi-manager #139: Custom Parameters

markirwin, 2021-08-22:
I would like to add a custom parameter on setup to configure an MQTT server and username / password. I do not see a way to do this with the current code. I want to save these parameters to NVS too and be able to use them in my code.

I am coming from moving code from arduino where there is a library that does this to idf native implementation.

I have thought if it were possible using a custom page but not sure if there is a way to make the login jump to this screen after configuring the wifi?

thebitguru, 2022-07-26:
@markirwin did you ever figure out how to accomplish this? I am in a similar situation was curious. Thanks!

markirwin, 2022-07-26:
@thebitguru, i did not and ended up implementing it using a config file. There is also lots of forks on this code with improvements, I would look at this for any help.

closed 2022-07-26 by markirwin

thebitguru, 2022-07-26:
Thanks for responding! I started looking at the forks and it seems that there are quite a few. Any specific one that you would recommend?