tonyp7/esp32-wifi-manager #141: DNS requests from Android11 are too long and cause disconnects

tofublock, 2021-08-30:
### Prerequisites

- [x] I have written a descriptive issue title
- [x] I have verified that I am using the latest version of esp32-wifi-manager
- [x] I have searched [open](https://github.com/tonyp7/esp32-wifi-manager/issues) and [closed](https://github.com/tonyp7/esp32-wifi-manager/issues?q=is%3Aissue+is%3Aclosed) issues to ensure it has not already been reported

### Description
Building this project out of the box and trying to connect to the captive portal from Android11 I get "Header fields are too long for server to interpret", which is fixed by adjusting `CONFIG_HTTPD_MAX_REQ_HDR_LEN=1024` in menuconfig (a lower value might work.)
Connecting to the device after adjusting this I see the captive portal, but am kicked off the connection within about 5 seconds almost every time. After a few tries it sometimes stays on.
I suspect that the problem is most likely with the DNS server. It seems that a lot of DNS requests from the connecting device bring the unwanted behaviour out more.
All this happens with no log output - just a WIFI_EVENT_AP_STADISCONNECTED.

Similar behaviour is seen from a Win10 device with Firefox, where a browser full of AJAX tabs will prevent the captive portal from loading, but without a browser open it comes up fine. In this case at least the connection is not interrupted and navigating to 10.10.0.1 works.

Has anybody else seen this behaviour and found a way around it? My next guess would be to start poking around in the DNS server response, but I'm a bit out of my depth here...

Regardless of that issue, thanks a lot for this project, @tonyp7!

### Steps to Reproduce
1. Build and flash master branch as is and try to connect from Android11 => "Header fields are too long for server to interpret"
2. Change `CONFIG_HTTPD_MAX_REQ_HDR_LEN=1024` => Captive portal comes up for a few seconds, then WiFi disconnects

### System Configuration
ESP-IDF 4.3
ESP32 chip revision 1

Vxs14, 2022-01-24:
Hy, i have got the same issue with ESP-IDF 4.2, did you find a solution ?

tofublock, 2022-01-24:
Turning off the DNS server helps, at the cost of losing captive portal functionality.

Vxs14, 2022-01-24:
Thank you, indeed it helps for the connection but I should like to use some potal functionalities of the portal, i will try to rewrite the DNS code

tofublock, 2022-01-24:
I found that only replying to a specific domain request helps a lot: Encapsulate Lines 159 to 171 of dns_server.c in an if(strcmp(domain, "portal.info") == 0) block.

I'd be curious to know if you figure out more about it!