tonyp7/esp32-wifi-manager #152: Support IDF v5.0

lida2003, 2022-12-21:
**Is your feature request related to a problem? Please describe.**
Support IDF v5.0 

**Describe the solution you'd like**
Support IDF   release/v5.0

```
commit 7f4bcc36959b1c483897d643036f847eb08d270e (HEAD -> release/v5.0, origin/release/v5.0)
Merge: 3f2f35bd5e f8d4bc8911
Author: Michael (XIAO Xufeng) <xiaoxufeng@espressif.com>
Date:   Sat Nov 12 00:28:57 2022 +0800

    Merge branch 'bugfix/revision_format_v5.0' into 'release/v5.0'

    soc: upgrade version printing to vX.Y (v5.0)

    See merge request espressif/esp-idf!21011

```

**Describe alternatives you've considered**
TBD

**Additional context**
```
esp32-wifi-manager/examples/default_demo$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/daniel/Work/esp32-wifi-manager/examples/default_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/daniel/Work/esp32-wifi-manager/examples/default_demo"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.25.1")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/daniel/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /home/daniel/Work/esp-idf-v5.0/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'mdns'.
Call Stack (most recent call first):
  /home/daniel/Work/esp-idf-v5.0/tools/cmake/build.cmake:281 (__build_resolve_and_add_req)
  /home/daniel/Work/esp-idf-v5.0/tools/cmake/build.cmake:574 (__build_expand_requirements)
  /home/daniel/Work/esp-idf-v5.0/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/daniel/Work/esp32-wifi-manager/examples/default_demo/build/CMakeFiles/CMakeOutput.log".
HINT: The component mdns has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details.
cmake failed with exit code 1, output of the command is in the /home/daniel/Work/esp32-wifi-manager/examples/default_demo/build/log/idf_py_stderr_output_10789 and /home/daniel/Work/esp32-wifi-manager/examples/default_demo/build/log/idf_py_stdout_output_10789

```

huardti, 2023-02-20:
the PR #153 can be use to fixe that !

CJCombrink, 2023-07-29:
I can confirm, pr#153 solves the issue for IDF v5.1
I don't see the issue as commented on the pr, it just works

to-sta, 2025-01-03:
I had an issue with the wifi types:

```
error: unknown type name 'wifi_bandwidth_t'
  253 |         wifi_bandwidth_t ap_bandwidth;
      |         ^~~~~~~~~~~~~~~~
...
```

Had to add the #include "esp_wifi_types.h" to the wifi_manger.h file.