tonyp7/esp32-wifi-manager #156: I was thinking a way to have ota to have integrate to wifi manager. is it possible as it seems a bit complicated with i getting E (4251) httpd: httpd_server_init: error in listen (112)

Shakir555, 2023-08-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Shakir555, 2023-08-23:
````c
// Libraries
#include <stdio.h>
#include <inttypes.h>
#include <string.h>
#include <esp_wifi.h>
#include <esp_netif.h>
#include <http_app.h>
#include <freertos/FreeRTOS.h>
#include <esp_http_server.h>
#include <freertos/task.h>
#include <esp_ota_ops.h>
#include <esp_system.h>
#include <lwip/ip4_addr.h>
#include <esp_log.h>
#include <nvs_flash.h>
#include <sys/param.h>

#include <wifi_manager.h>

#define WIFI_SSID "ESP32 OTA Update"

// Tag used for ESP Serial Console Messages
static const char TAG[] = "main";

void monitoring_task(void *pvParameter) {
    for (;;) {
        ESP_LOGI(TAG, "free heap: %"PRIu32, esp_get_free_heap_size());
        vTaskDelay(pdMS_TO_TICKS(10000));
    }
}

/*
 * Serve OTA update portal (index.html)
 */
extern const uint8_t index_html_start[] asm("_binary_index_html_start");
extern const uint8_t index_html_end[] asm("_binary_index_html_end");

esp_err_t index_get_handler(httpd_req_t *req)
{
	httpd_resp_send(req, (const char *) index_html_start, index_html_end - index_html_start);
	return ESP_OK;
}

esp_err_t update_post_handler(httpd_req_t *req)
{
	char buf[2000];
	esp_ota_handle_t ota_handle;
	int remaining = req->content_len;

	const esp_partition_t *ota_partition = esp_ota_get_next_update_partition(NULL);
	ESP_ERROR_CHECK(esp_ota_begin(ota_partition, OTA_SIZE_UNKNOWN, &ota_handle));

	while (remaining > 0) {
		int recv_len = httpd_req_recv(req, buf, MIN(remaining, sizeof(buf)));

		// Timeout Error: Just retry
		if (recv_len == HTTPD_SOCK_ERR_TIMEOUT) {
			continue;

		// Serious Error: Abort OTA
		} else if (recv_len <= 0) {
			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Protocol Error");
			return ESP_FAIL;
		}

		// Successful Upload: Flash firmware chunk
		if (esp_ota_write(ota_handle, (const void *)buf, recv_len) != ESP_OK) {
			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Flash Error");
			return ESP_FAIL;
		}

		remaining -= recv_len;
	}

	// Validate and switch to new OTA image and reboot
	if (esp_ota_end(ota_handle) != ESP_OK || esp_ota_set_boot_partition(ota_partition) != ESP_OK) {
			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Validation / Activation Error");
			return ESP_FAIL;
	}

	httpd_resp_sendstr(req, "Firmware update complete, rebooting now!\n");

	vTaskDelay(500 / portTICK_PERIOD_MS);
	esp_restart();

	return ESP_OK;
}

/*
 * HTTP Server
 */
httpd_uri_t index_get = {
	.uri	  = "/",
	.method   = HTTP_GET,
	.handler  = index_get_handler,
	.user_ctx = NULL
};

httpd_uri_t update_post = {
	.uri	  = "/update",
	.method   = HTTP_POST,
	.handler  = update_post_handler,
	.user_ctx = NULL
};

static esp_err_t http_server_init(void)
{
	static httpd_handle_t http_server = NULL;

	httpd_config_t config = HTTPD_DEFAULT_CONFIG();

	if (httpd_start(&http_server, &config) == ESP_OK) {
		httpd_register_uri_handler(http_server, &index_get);
		httpd_register_uri_handler(http_server, &update_post);
	}

	return http_server == NULL ? ESP_FAIL : ESP_OK;
}

void ota_update_task(void *pvParameters) {
    // Mark current app as valid
    const esp_partition_t *partition = esp_ota_get_running_partition();
    ESP_LOGI(TAG, "Currently running partition:%s\r\n", partition->label);

    esp_ota_img_states_t ota_state;
    if (esp_ota_get_state_partition(partition, &ota_state) == ESP_OK) {
        if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
            esp_ota_mark_app_valid_cancel_rollback();
        }
    }
    for (;;) {
        ESP_LOGI(TAG, "OTA Running");
        vTaskDelay(pdMS_TO_TICKS(10000));
    }
}

void ota_update_task_create(void) {

	http_server_init();

    // Create the OTA updates task
    xTaskCreatePinnedToCore(&ota_update_task, "ota task", 4056, NULL, 1, NULL, 1);
}

void cb_connection_ok(void *pvParameter) {

	ip_event_got_ip_t *param = (ip_event_got_ip_t *)pvParameter;

	//Transform IP to human readable string
	char str_ip[16];
	esp_ip4addr_ntoa(&param->ip_info.ip, str_ip, IP4ADDR_STRLEN_MAX);

	esp_ip4_addr_t zero_ip;
	memset(&zero_ip, 0, sizeof(esp_ip4_addr_t));

	if (memcmp(&param->ip_info.ip, &zero_ip, sizeof(esp_ip4_addr_t)) != 0) {
		ota_update_task_create();
	}
}

void app_main(void) {
    esp_err_t ret = nvs_flash_init();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }

    ESP_ERROR_CHECK(ret);

    // Start the Wifi Manager
    wifi_manager_start();

    // Register a callback to integrate code with the wifi manager
    wifi_manager_set_callback(WM_EVENT_STA_GOT_IP, &cb_connection_ok);

    // Here, create a task on core 1 that monitors free heap memory
    xTaskCreatePinnedToCore(&monitoring_task, "monitoring task", 4056, NULL, 1, NULL, 1);
}
```

Shakir555, 2023-08-23:
///httpd_app.c

/*
Copyright (c) 2017-2020 Tony Pottier

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.

@file http_app.c
@author Tony Pottier
@brief Defines all functions necessary for the HTTP server to run.

Contains the freeRTOS task for the HTTP listener and all necessary support
function to process requests, decode URLs, serve files, etc. etc.

@note http_server task cannot run without the wifi_manager task!
@see https://idyl.io
@see https://github.com/tonyp7/esp32-wifi-manager
*/

#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <stdint.h>
#include <esp_wifi.h>
#include <esp_event.h>
#include <esp_log.h>
#include <esp_system.h>
#include <esp_netif.h>
#include <esp_http_server.h>

#include <wifi_manager.h>
#include <http_app.h>

/* @brief tag used for ESP serial console messages */
static const char TAG[] = "http_server";

/* @brief the HTTP server handle */
static httpd_handle_t httpd_handle = NULL;

/* function pointers to URI handlers that can be user made */
esp_err_t (*custom_get_httpd_uri_handler)(httpd_req_t *r) = NULL;
esp_err_t (*custom_post_httpd_uri_handler)(httpd_req_t *r) = NULL;

/* strings holding the URLs of the wifi manager */
static char* http_root_url = NULL;
static char* http_redirect_url = NULL;
static char* http_js_url = NULL;
static char* http_css_url = NULL;
static char* http_connect_url = NULL;
static char* http_ap_url = NULL;
static char* http_status_url = NULL;

/**
 * @brief embedded binary data.
[... 371 lines cut ...]
			}
			else{
				snprintf(http_redirect_url, redirect_sz, "http://%s%s", DEFAULT_AP_IP, WEBAPP_LOCATION);
			}

			/* generate the other pages URLs*/
			http_js_url = http_app_generate_url(page_js);
			http_css_url = http_app_generate_url(page_css);
			http_connect_url = http_app_generate_url(page_connect);
			http_ap_url = http_app_generate_url(page_ap);
			http_status_url = http_app_generate_url(page_status);

		}

		err = httpd_start(&httpd_handle, &config);

	    if (err == ESP_OK) {
	        ESP_LOGI(TAG, "Registering URI handlers");
	        httpd_register_uri_handler(httpd_handle, &http_server_get_request);
	        httpd_register_uri_handler(httpd_handle, &http_server_post_request);
	        httpd_register_uri_handler(httpd_handle, &http_server_delete_request);
	    }
	}

}

CJCombrink, 2023-08-23:
Hi @Shakir555 

### First off
Just some general guidance on posting in GitHub in general (or any other project) 
1. Keep the title short and on point
    It helps the owner to see at a glance what the ticket is about
1. Add the needed information in the body of the new template.
1. Make sure to follow the template presented when you "Create a ticket".
    Fill out all information that is requested in full.
    If you think some information that is requested is not relevant, always state why you think it is not relevant, don't just remove  or leave it empty.
 
For this specific issue, the code added to the ticket has no context and there is no meaningful way to assist. 
If you change existing code and need help, fork the repo and make your changes and make a PR. That way one can assist by looking at the changes you have made.

Please, as a first step, reword your title and edit the first comment and complete the information there as required. 

### Trying to answer
Ok, sorry I had to state those, now to try and assist:
I am successfully using this repo along with OTA.
I am not sure where you are struggling with, but I could plugin in all of my OTA logic on top of this library by using the [http_hook](https://github.com/tonyp7/esp32-wifi-manager/tree/master/examples/http_hook) example as-is.
Look at the [my_get_handler()](https://github.com/tonyp7/esp32-wifi-manager/blob/master/examples/http_hook/main/user_main.c#L44)

PS: I am not associated with this repo, I just find it useful. I will most likely close an issue like this if it pops up on my repo.

CJCombrink, 2023-08-23:
A quick glance at the code, you most certainly should not try to start a new web server.
This line makes me think you are trying to: 
```
err = httpd_start(&httpd_handle, &config);
```

You have to plug into the web server started by the wifi manager (this repo) and then handle the request you get, as per the function I linked above.

Shakir555, 2023-08-25:
@CJCombrink Hi CJ thanksso much. any ideas how to do the fork the repo and make your changes and make a PR. Im so sorry because Im very novice to programming, github etc....

Shakir555, 2023-08-25:
Hi I am back, here my new issue still got problem with error 404.
Below is what i working on.
[image]
[image]

This my code below 

```c

// Libraries
#include <stdio.h>
#include <inttypes.h>
#include <string.h>
#include <esp_wifi.h>
#include <esp_netif.h>
#include <http_app.h>
#include <freertos/FreeRTOS.h>
#include <esp_http_server.h>
#include <freertos/task.h>
#include <esp_ota_ops.h>
#include <esp_system.h>
#include <lwip/ip4_addr.h>
#include <esp_log.h>
#include <nvs_flash.h>
#include <sys/param.h>

#include <wifi_manager.h>

/* @brief tag used for ESP serial console messages */
static const char TAG[] = "main";

void monitoring_task(void *pvParameter) {
    for (;;) {
        ESP_LOGI(TAG, "free heap: %"PRIu32, esp_get_free_heap_size());
        vTaskDelay(pdMS_TO_TICKS(10000));
    }
}

/*
 * Serve OTA update portal (index.html)
 */
extern const uint8_t ota_html_start[] asm("_binary_ota_html_start");
extern const uint8_t ota_html_end[] asm("_binary_ota_html_end");

static esp_err_t my_get_handler(httpd_req_t *req){

	/* our custom page sits at /helloworld in this example */
	if(strcmp(req->uri, "/ota") == 0){

		ESP_LOGI(TAG, "Serving page /ota");
		httpd_resp_send(req, (const char *) ota_html_start, ota_html_end - ota_html_start);

		httpd_resp_set_status(req, "200 OK");
		httpd_resp_set_type(req, "text/html");
		
	}
	else{
		/* send a 404 otherwise */
		httpd_resp_send_404(req);
	}

	return ESP_OK;
}

esp_err_t update_post_handler(httpd_req_t *req)
{
	char buf[1000];
	esp_ota_handle_t ota_handle;
	int remaining = req->content_len;

	const esp_partition_t *ota_partition = esp_ota_get_next_update_partition(NULL);
	ESP_ERROR_CHECK(esp_ota_begin(ota_partition, OTA_SIZE_UNKNOWN, &ota_handle));

	while (remaining > 0) {
		int recv_len = httpd_req_recv(req, buf, MIN(remaining, sizeof(buf)));

		// Timeout Error: Just retry
		if (recv_len == HTTPD_SOCK_ERR_TIMEOUT) {
			continue;

		// Serious Error: Abort OTA
		} else if (recv_len <= 0) {
			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Protocol Error");
			return ESP_FAIL;
		}

		// Successful Upload: Flash firmware chunk
		if (esp_ota_write(ota_handle, (const void *)buf, recv_len) != ESP_OK) {
			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Flash Error");
			return ESP_FAIL;
		}

		remaining -= recv_len;
	}

	// Validate and switch to new OTA image and reboot
	if (esp_ota_end(ota_handle) != ESP_OK || esp_ota_set_boot_partition(ota_partition) != ESP_OK) {
			httpd_resp_send_err(req, HTTPD_500_INTERNAL_SERVER_ERROR, "Validation / Activation Error");
			return ESP_FAIL;
	}

	httpd_resp_sendstr(req, "Firmware update complete, rebooting now!\n");

	vTaskDelay(500 / portTICK_PERIOD_MS);
	esp_restart();

	return ESP_OK;
}

httpd_uri_t update_post = {
	.uri	  = "/ota",
	.method   = HTTP_POST,
	.handler  = update_post_handler,
	.user_ctx = NULL
};

void ota_update_task(void *pvParameters) {
    // Mark current app as valid
    const esp_partition_t *partition = esp_ota_get_running_partition();
    ESP_LOGI(TAG, "Currently running partition:%s\r\n", partition->label);

    esp_ota_img_states_t ota_state;
    if (esp_ota_get_state_partition(partition, &ota_state) == ESP_OK) {
        if (ota_state == ESP_OTA_IMG_PENDING_VERIFY) {
            esp_ota_mark_app_valid_cancel_rollback();
        }
    }
    for (;;) {
        ESP_LOGI(TAG, "OTA Running");
        vTaskDelay(pdMS_TO_TICKS(10000));
    }
}

void ota_update_task_create(void) {
    // Create the OTA updates task
    xTaskCreatePinnedToCore(&ota_update_task, "ota task", 4056, NULL, 1, NULL, 1);
}

void cb_connection_ok(void *pvParameter) {

	ip_event_got_ip_t *param = (ip_event_got_ip_t *)pvParameter;

	//Transform IP to human readable string
	char str_ip[16];
	esp_ip4addr_ntoa(&param->ip_info.ip, str_ip, IP4ADDR_STRLEN_MAX);

	esp_ip4_addr_t zero_ip;
	memset(&zero_ip, 0, sizeof(esp_ip4_addr_t));

	if (memcmp(&param->ip_info.ip, &zero_ip, sizeof(esp_ip4_addr_t)) != 0) {
		ota_update_task_create();
	}
}

void app_main()
{
	/* start the wifi manager */
	wifi_manager_start();

	// Register a callback to integrate code with the wifi manager
    wifi_manager_set_callback(WM_EVENT_STA_GOT_IP, &cb_connection_ok);

	/* set custom handler for the http server
	 * Now navigate to /helloworld to see the custom page
	 * */
	http_app_set_handler_hook(HTTP_GET, &my_get_handler);
}

```

CJCombrink, 2023-08-26:
@Shakir555 Thanks for updating the comment, now please update the header, something like: "Need help - problem getting OTA to work with Wifi Manager"

> any ideas how to do the fork the repo and make your changes and make a PR.

For your specific issue, you don't need a PR since you can use this repo (Wifi Manager) as-is in your application, you don't need to modify it. A PR is for when you are trying to make or suggest changes to someone else's repo.

>  here my new issue still got problem with error 404.

Ok, that issue looks familiar. If I look at your code I am going to guess it is this part: 

```
	httpd_resp_sendstr(req, "Firmware update complete, rebooting now!\n");
	vTaskDelay(500 / portTICK_PERIOD_MS);
	esp_restart();
```

If I look at the screenshot it appears that the upload worked (reported `100%` progress)
Can you confirm that the OTA did work? Did your device at least restart and applied the new OTA?
I am assuming yes.

The issue is then that you reboot the ESP before the ESP gets a chance to send the last response to the device. The solution is to move the reboot to another task. 
Look at this example: https://github.com/backinabit/ESP32-OTA-Webserver/blob/master/OTAServer.c
Specifically the [systemRebootTask](https://github.com/backinabit/ESP32-OTA-Webserver/blob/master/OTAServer.c#L36C12-L36C12)
and where it is called from [OTA_update_status_handler](https://github.com/backinabit/ESP32-OTA-Webserver/blob/master/OTAServer.c#L99)

PS: The [backinabit/ESP32-OTA-Webserver](https://github.com/backinabit/ESP32-OTA-Webserver) repo is almost exactly the code I use for my OTA with this repo and it works without issues.

Shakir555, 2023-08-28:
@CJCombrink Hi CJ, I already changes the header. I try following the ota repo that you recommended. try to mix out the code. but still seems no luck on it on the 404 error. Below is my main code and my html code. i quite suspect my html code somehow prevent the OTA to update on the esp32. 

**user_main.c**

```c

//Libraries
#include <stdio.h>
#include <stdlib.h>
#include <inttypes.h>
#include <string.h>
#include <esp_wifi.h>
#include <esp_netif.h>
#include <http_app.h>
#include <freertos/FreeRTOS.h>
#include <esp_http_server.h>
#include <freertos/task.h>
#include <freertos/event_groups.h>
#include <esp_ota_ops.h>
#include <esp_system.h>
#include <lwip/ip4_addr.h>
#include <esp_log.h>
#include <nvs_flash.h>
#include <sys/param.h>

#include <wifi_manager.h>

static const char TAG[] = "main";

//Serve OTA update portal (index.html)
extern const uint8_t index_html_start[] asm("_binary_index_html_start");
extern const uint8_t index_html_end[] asm("_binary_index_html_end");

//Variables
httpd_handle_t ota_server = NULL;
int8_t flash_status = 0;

EventGroupHandle_t reboot_event_group;
const int REBOOT_BIT = BIT0;

//System Reboot Task
void systemRebootTask(void *pvParameter)
{
	//Init the event group
	reboot_event_group = xEventGroupCreate();

	//Clear the bit
	xEventGroupClearBits(reboot_event_group, REBOOT_BIT);

	for (;;)
	{
		//Wait here until the bits gets set for reboot
		EventBits_t staBits = xEventGroupWaitBits(reboot_event_group, REBOOT_BIT, pdTRUE, pdFALSE, portMAX_DELAY);

		//Did portMAX_DELAY evet timeout, not sure so let just check to be sure
		if ((staBits & REBOOT_BIT) != 0)
		{
			ESP_LOGI(TAG, "Reboot command, Restarting");
			vTaskDelay(2000 / portTICK_PERIOD_MS);

			esp_restart();
		}
	}
}

//Send index_html Page
esp_err_t ota_index_html_handler(httpd_req_t *req)
{
	ESP_LOGI(TAG, "index_html Requested");

[... 388 lines cut ...]
						} else if (xhr.status == 0) {
							alert("Server closed the connection abruptly!");
							location.reload()
						} else {
							alert(xhr.status + " Error!\n" + xhr.responseText);
							location.reload()
						}
					}
				};
					

            xhr.upload.onprogress = function(e) {
                var progress = document.getElementById("progress");
                progress.textContent =
                    "Progress: " + ((e.loaded / e.total) * 100).toFixed(0) + "%";
            };
            xhr.open("POST", "/update", true);
            xhr.send(file);
            xhr.close("Done");
        }
    }
</script>
</body>
</html>
```

Shakir555, 2023-08-28:
When I try to Post/Update. it got this Warning/Error W (23504) httpd_txrx: httpd_sock_err: error in recv : 104
[image]

Shakir555, 2023-08-28:
@CJCombrink Hi CJ, Okay Im back, I already solved the issue following your repo using the http_app_set_handler_hook on the void app_main(). Copy some of the important OTA src code and put it in the src file in the WM.  It works good. Thanks buddy

```c
        http_app_set_handler_hook(HTTP_GET, &OTA_html_handler);
	http_app_set_handler_hook(HTTP_POST, &OTA_update_post_handler);
	http_app_set_handler_hook(HTTP_POST, &OTA_update_status_handler);
```
[image]

closed 2023-09-08 by Shakir555